opentx/opentx #5215: Inputs in a Lua Script

Storchy1, 2017-09-15:
i have tried to make 6 input values in a lua script OpenTx 2.2
local options = {
  { "Pitch" , VALUE,  2,-100,100 },
  { "Head"  , VALUE, 70,   0,100 },
  { "NRExpo", VALUE, 25,   0,100 },
  { "HExpo" , VALUE, 25,   0,100 },
  { "NRDuRa", VALUE, 75,  25,100 },
  { "HDuRa" , VALUE, 75,  25,100 }
}
i cannot get the last value HDuRa, it is always nil.
I thought it is possible to use 6 input values.
Please help

projectkk2glider, 2017-09-15:
Should work with 6 inputs. Can you post a minimalist Lua code that exhibits your problem.

kilrah, 2017-09-15:
Max number of options is 5.

Storchy1, 2017-09-16:
Thanks for your help!.
@kilrah : You write 5 options for Horus, please can you tell me where i can read this, because i have read that the options in 2.2 are 6. I there an extra documentation for Horus. I have another problem that i have tried to set the BLINK option and nothing happend!

I have a Horus with OpenTX 2.2
I try to add the lua script.
[main.zip](https://github.com/opentx/opentx/files/1308468/main.zip)

Thanks a lot, you all make a really good job.

Storchy1, 2017-09-16:
I have tried the widget in the simulator.
I can only see the first 5 inputs, perhaps that helps.
[image]

3djc, 2017-09-16:
As Kilrah said, max for horus widgets is 5

projectkk2glider, 2017-09-16:
Widgets have **options** - max 5 of them. Mixer scripts have **inputs** - max 6 of them. The docs for Widgets and Themes are still missing https://github.com/opentx/lua-reference-guide/issues/30.

closed 2017-09-16 by projectkk2glider