opentx/opentx #5216: Channels Monitor not updated in Simulator: X9D+

powercycle, 2017-09-15:
When starting fresh and creating a new model, Channels Monitor does not display stick movement from within the simulator when started from model editor.

Additionally, no warnings are displayed when simulating from within model editor.

Works fine when starting simulator directly from main screen.

Steps to reproduce are:
- start new otx file
- start new model 1
- Cancel wizard
- edit model
- simulate radio
- page button until channel monitors displays
- move sticks, no display

steps to fix the problem:
- simulate radio from main menu (not model editor)
- page button until channel monitors displays
- move sticks, display works (and you get warnings)

After this, going back to model editor works as expected, however, warnings are still not displayed.

This may be a duplicate of https://github.com/opentx/opentx/issues/4529, but I'm not completely sure.

This is on Ubuntu 16 using the PPA on version 2.2.0, May 31 2017

projectkk2glider, 2017-09-15:
The startup **warnings are skipped** when simulating from model editor - this is by design!

For the channel monitor not moving, what do you see in mixes/inputs in the same simulator session, perhaps they are empty - ie no default mixes.

powercycle, 2017-09-15:
The mixes and inputs are there, I did check. 
As far as the warnings being skipped by design, thank you for the information.

powercycle, 2017-09-16:
Just for fun I tried it again.  Now I can't get it to fail.   That is very strange.

This was repeatable over a couple of tries before I created the ticket.  I'm going to close it for now until I can come up with an absolute way to reproduce the issue.  It is very strange.

projectkk2glider, 2017-09-16:
>It is very strange.

Indeed...

kilrah, 2017-09-16:
The simu inputs occasionally not working has been reported by a few other people as well, usually it starts working again after a couple of simu reloads. It was already noticed during the RC period, @mpaperno had investigated and made some fixes that made it better but it's still happening sometimes for some.

hrenz, 2017-09-16:
Same at my PC Win10 Prof with companion V2.20, with X9E, X9DP, X7D
I have made a video
sometimes it works at first time start of the simulator ok
sometimes after 1 or 2 reload, sometimes not after x tries
[Video_2017-09-11_195144.zip](https://github.com/opentx/opentx/files/1308496/Video_2017-09-11_195144.zip)

kilrah, 2017-11-08:
Duplicate of #4529

closed 2017-11-08 by kilrah