opentx/opentx #5217: The ordinal 3873 could not be located in the dynamic link library ... when starting companion 2.2.0

dlondon2017, 2017-09-16:
Hi,

Just installed Companion 2,2,0 on Windows 10

I get popup errors when I start companion:

The ordinal 3873 could not be located in the dynamic link library C:\Program Files (x86)\iis express\PHP\v5.4\ssleay32.dll
then
The ordinal 3873 could not be located in the dynamic link library C:\Program Files (x86)\PHP\v5.4\ssleay32.dll

Any ideas on how to fix?

Regards,

David

projectkk2glider, 2017-09-16:
If we are using OpenSSL library, that dll should be included into installation dir. This way the system will not use other ssleay32.dll which in this case is apparently not-compatible - ie wrong verson.

@kilrah and @3djc  you are more familiar with windows build, any clue.

3djc, 2017-09-16:
We have add a couple of those; I think that's conflicting ssl stacks on the PC. I thought that @mpaperno made a change to have it incorporated

kilrah, 2017-09-16:
I think there were talks of it but nothing actually done. No problem to do it, question is what version should we bundle and where to source it from for "build environment friendliness"...

dlondon2017, 2017-09-17:
Thanks for looking into this.  I did try downloading openssl-1.0.2l-x64_86-win64.zip and placing ssleay32.dll in C:\Program Files (x86)\OpenTX\Companion 2.2 and also C:\Windows\System32 but I guess those crude attempts were not enough because the same errors came up

Any idea if Companion will work  despite this error message?

Regards,

David

projectkk2glider, 2017-09-17:
I can't find why we even need SSL lib. Anyone know? 

Looking at the cmake we only try to include that ddl in mingw build:
```
  elseif(MINGW)
    get_filename_component(MINGW_DIR ${CMAKE_CXX_COMPILER} PATH)
    if(IS_DIRECTORY ${MINGW_DIR})
      install(FILES "${MINGW_DIR}/libgcc_s_dw2-1.dll" "${MINGW_DIR}/libstdc++-6.dll" "${MINGW_DIR}/libwinpthread-1.dll" DESTINATION ${INSTALL_DESTINATION})
      set(MINGW_DIR "${MINGW_DIR}/../opt/bin")
      # SSL support
      install(FILES "${MINGW_DIR}/libeay32.dll" "${MINGW_DIR}/ssleay32.dll" DESTINATION ${INSTALL_DESTINATION} OPTIONAL)
    endif()
```

MSVC build does not have this. This was introduced in e90bd0fd706dee3abc02320ee7107801e5d75243. We did not include that ddl before.

3djc, 2017-09-17:
I think the ssl lib is needed for file download from download.opentx,

projectkk2glider, 2017-09-17:
I just checked dependencies of Companion on Linux, no libssl needed. Weird.

projectkk2glider, 2017-09-17:
Looks like Linux version of QT has ssl embedded while the Windows version requires separate dlls https://stackoverflow.com/questions/20351155/how-can-i-enable-ssl-in-qt-windows-application

3djc, 2017-09-17:
We do not use https downloads, but my undersatnding (which most definitively  can be wrong) is the software bit from qt that we use is able to handle both, and therefore requires ssl

kilrah, 2017-09-17:
I think we had figured out that we don't need or use OpenSSL at all, but the version of QT5Network we have tries to load it regardless of whether it's actually needed or not. If it's not found it's not a problem and nothing happens, but if the system has a copy in the PATH it loads it and barfs that error if it's an unexpected version.

If everything works well you should be fine.

projectkk2glider, 2017-09-28:
Since we now use https, this needs fixing before the 2.2.1 release!

pboschi, 2017-10-07:
Unpack this into Companion directory until fixed:
https://indy.fulgan.com/SSL/openssl-1.0.2l-i386-win32.zip

That binary distribution is suggested by:
https://wiki.openssl.org/index.php/Binaries

dlondon2017, 2017-10-07:
Thanks!  That did it.  Looks like only the DLL's are needed: libeay32.dll and ssleay32.dll.  The executable openssl.exe doesn't get used which makes sense.

Thanks again for you help

dlondon2017, 2017-10-07:
whoops - new to GitHub - didn't realise what close and comment meant.  I'll leave it to the Developers to decide when to close

commit 6a14fe72 (2017-10-08): Bundle OpenSSL libs in Windows installer (fix #5217) and image format plugins the app loads
closed 2017-10-08 by kilrah