opentx/opentx #5222: sound file not closed by edit special functions module?

Smoothead, 2017-09-17:
Companion 2.2 
After adding a 'play track' sound in special functions
Press the test play button.  Sound is played.
Now simulate. Sound is not played in response to triggering switch.
Log says file access denied.
I suspect this is because the file is not closed after doing a test play in the special functions window.
If Companion is closed and restarted, model opened, simulate, sound is played.
If sound is played again in the special functions window, once again, the simulator fails to play sound.
Screen grab to illustrate
[image]

The error was reported to me by a member of our club who uses Windows 7 - 
I am seeing the same results using Windows 10.  Both of us using May 30th 2.2
I wonder if other can confirm that there is a bug or offer a solution?

mpaperno, 2018-06-05:
Related: #4929

closed 2018-06-29 by bsongis

Smoothead, 2018-06-29:
Thank you, I look forward to the new version with the correction applied.

PR #6000 (2018-06-29): Companion custom functions sound playback fixes
* Fix sound file locked after playback (fixes #5222, fixes #5997); 
* Refactor play button (fixes #4929);
* Fix possible crash when sound file locked from outside (eg. SDL in simu);
* Adds better feedback (error msg) when a file can't be opened (eg. locked or unreadable);

Tested in all supported OSs/compilers.

I did discover another related bug, this one on the simu firmware side, where it (I'm assuming SDL) doesn't close a file when it is used as background music, or if a file couldn't be played back for some other reason.  So the file remains [... cut]
commit 2eec091a (2018-06-29): Companion custom functions sound playback fixes (#6000)

* [CompanionIcon] Can now add multiple image files to icon for representing different UI states (checked/active/etc).

* [Companion][CustomFunctionsPanel] Fix sound file locked after playback (fixes #5222, fixes #5997); Refactor play button (fixes #4929); Create new QMediaPlayer on each play and load in "LowLatency" mode instead of stream.