opentx/opentx #5223: Add to Tx setup selection for RSSI threshold values

franz4949, 2017-09-18:
To date the threshold values for RSSI warnings can be modified on last page of model menu and preselection of values is set to 45 / 42 dB. Myself and as I know also other users lower that values for their models, for me I put the values down each for 3 dB, thus to 42 / 39 dB.
It would be more comfortable, if those preselected threshold values could be defined in general somewhere in the Tx setup, thus make it unnecessary to do this for each model separately. Although the option for such separate modification should also be maintained for each model in the model menu page (where it is now).

3djc, 2017-09-18:
You can simply define a model to be copied that includes all your basic preferences, that's what most do

franz4949, 2017-09-18:
Yes, I know I can do that. My request or proposal is just for additional comfort and ofcourse not essential and of lowest priority (if decided for implementation in a future release). I also have no problem to close it if you want.

powercycle, 2017-09-18:
I'm curious as to what made the OpenTX team choose the default values that are currently there?

kilrah, 2017-09-18:
They were the values used by FrSky on their own equipment before we came into play.

Devil-67, 2017-09-18:
Hi,

I would be very happy if it would be possible,
which can set the user at the RSSI warning threshold a delay, without defining logic switches.

Reason:
There are in the 2.4GHZ much short peaks a few milliseconds which immediately lead to a warning.
The nerft and newcomer get insecure.

My personal solution is I reduce the original limits to 20 and define new logic switches with delay.
This costs space and is actually unnecessary and cumbersome.
[image]

[image]

[image]

It is not a real problem, but it would make much easier,
  even for newcomers who do not know this.

I would be happy if you think about it ... ;-)
And thanks for this great achievement you created with OpenTX.

schwabe, 2017-09-18:
We implmenent something like a new default or so, I think we are going to implement something that works for various setting and not just one. Otherwise it is going to be a mess

Devil-67, 2017-09-18:
You since the programmers, it was only an idea since it was a long time nerve, look times in the FPV again and again these questions also dive:
I have poor reception, the Taranis signal badly ....

I would find the super, look at the screenshot ;-))

[image]

franz4949, 2017-09-18:
I also think, that besides my proposal adding a delay option is worth to be considered (as proposed by Devil-67) where a delay range (in tenth of a second from 0 to 10 and a preselected set value of maybe 2 or 3) can be set before a threshold warning becomes active. As Devil already wrote, by such delay an alarm just by any short low RSSI peak would be avoided.

3djc, 2017-09-19:
That's all interesting, but it seems to not take into account that RSSI is computed by receiver and sent though telemetry to the radio, And that data is sent about 2 times per second only (arround 440ms usualy)

franz4949, 2017-09-19:
No need to change FrSky's computing of data and yes, you are right with the mentioned update cycle of about 2 times per second. I have looked into some of my logfiles which I usuallly log with a frequency of 0.2 sec and I can find that such peaks mostly show 2 times same value, which quite correlates with your mentioned 440ms data update rate. Thus one option would be to select whether a delay (of 0.5s) shall be applied or not applied or to offer a selection for optional delay with 0s (no delay), 0.5s or 1.0s
I think that the first option with delay yes/no is the easiest one and should be sufficient in most cases.

Devil-67, 2017-09-19:
@ 3djc I 

Sorry for my English ...
I think I have expressed myself badly.
[image]

The RSSI value is OK and should not be changed!

Please look at times a few posts ahead of the screenshots of me.
Then you better understand what we mean.

I try to explain it again:

In Open TX, there are RSSI warning thresholds 45 and 42, if less than 45 dbi comes a warning, signal weak, but this is usually not so there are always only very short peaks that are below the limit.
New users are frightened by the warning and are looking for a hardware error.
This was with me 3 years ago synonymous so, I have antenne rebuilt etc ...
  but there are always warnings, until I saw the log
have seen and everything is ok, just a short peak.

Just like here:

my solution looks like this:
[image]
[image]
[image]

and so would be my wish:
Small, compact and understandable
[image]

3djc, 2017-09-19:
What you call 'short peak' cannot be less than half a second,  that's nowhere near the 'few milliseconds' you are suggesting. And if my memory is right, I think it takes 1 second for the receiver to go in failsafe, so ignoring one value, you could be in failsafe comes the next one

schwabe, 2017-09-19:
@Devil-67 Most users don't get these short peaks. This mean either you are really on the border of loosing reception there or FrSky's way of reporting rssi is broken. If I assume that the rssi reported from the Receiver is an average (since minimum does not make sense since there is always fading/lost frames etc), that would still mean that you had half a second of very low reception. I hesitate to implement something that turn out later to mask real problems.

kilrah, 2017-09-19:
Especially given that those who really want something like this already have a way of doing it.

Devil-67, 2017-09-19:
OK,
I am silent then ;-)

Perhaps even more users who also see it so,
there is already enough in FPV and at Engel MT Forum.

Thank you for your answers

franz4949, 2017-09-19:
Yes, we already agreed that there is a way to do it, no question. My initial request/proposal was only to adjust a little bit more comfortable the RSSI warning threshold values as standard for each new model created with the model wizard.

Regarding the additional wish for implementation of an optional delay, proposed by Devil-67, let me say following. First of all I would like to confirm that myself and also my club colleagues all fly within LOS. But with gliders (wingspan 3m .... 5m) we often go to visual sight limits chasing thermal lifts. I agree that possibly the majority of pilots never go so far but stay with their planes within 200-300m radius and heights less than 300m. But even then, if Rx antennas are unfavorable placed and become shadowed for a short turn of the model during flight, an RSSI drop below alarm threshold of the standard value of 45 may occur. I saw it on several occasions when the colleagues contacted me for that reason of getting an RSSI alarm. I could help them by firstly improving the placiement of the Rx antennas and secondly by lowering the two standard threshold alarm values by 3dB. So from my experience I cannot say that getting alarms always is either being on the border of loosing reception, nor is it that FrSky's way of reporting is broken. And as you can see from the occurrence and duration of logged low RSSI peaks, the next second everything is again ok. So the model was/is not at all in a severe critical flight condition close to loosing control but only in a short unfavorable flight attitude with most probably shadowed antennas.

And the proposal for optionally added delay (I proposed a simple selection delay yes/no with either zero or 0.5s delay) will to my understanding not mask really receiption problems, it will just block uncritical alarms caused otherwise by single momentary signal drop events.

But as also written earlier, if you consider this not worth to implement (because there are already options to do it) or if you still see a general risk for flying then I can close this request. I think we have exchanged our thoughts so far, so please let me know how you wish to go on with it.

Anyway many thanks for all of your engagement in this awesome project.

schwabe, 2017-09-19:
@franz4949 The RSSI values and how these are measured are simply outside the control of OpenTX. As these are critical for the safety of your model this is one of the places where we rather stick whatever Frsky tells us are the limits of their equipment rather then you second guess their decisions. If Frsky decides to update/change their RSSI calculation or thresholds for warning/critical we will update ours. Personal experiences might differ but note that we have to play safe here as OpenTX project.

franz4949, 2017-09-19:
Ok, I understand your position and concerns regarding the wish for a delay option.

So what is left is my initial request for putting an additional setup option for the RSSI threshold values into the Tx setup menu. Preselected values remain with 45/42 dB but can be modified there by user and will be applied for a new model when setup with the model wizard.

franz4949, 2017-09-20:
To avoid any misunderstanding, the threshold values 45/42 as specified by FrSky shall be maintained as preselected, only the access for a change is proposed to make additionally possible in the Tx setup menu.
Same with the discussed delay option as a no/yes selection with then 0/0.5s and 0s as preselected value.. Both can already be achieved by the user if he wants, only difference is that with actual programming options it is somewhat more complex to do and with proposed way it is easier.

Anyway I will close this item, since I obviously cannot convince someone of you that adding proposed feature does not create new risk for flying but only adds some comfort for programming. Thanks to all for the discussion.

closed 2017-09-20 by franz4949