opentx/opentx #5224: OpenTX Companion Radio Simulator telemetry issues

teckel12, 2017-09-18:
Running v2.2.0 on Taranis Q X7.  When loading a log file and simulating the telemetry I'm having a couple problems.

- GPS data shows as a value like `0.7264` and `-0.6935` instead of what's in the log file (in degrees).  The latitude appears to need a conversion to degrees from radians, but the longitude is way off even with this conversion (by a magnitude of 2.1 by my estimations).
- Sensor `A4` while 2 digits in the logs and setup within telemetry, is only an integer in the telemetry simulator.  I'm using `A4` as the cell voltage so it only every reports values of `3` and `4`, which doesn't work well for telemetry lua scripts that monitor low voltage. iNav uses `A4` for cell voltage.

I apologize if these have already been reported or addressed. Is there a development branch of OpenTX along with compile/build instructions?  As a programmer I could look into fixing these issues if there's some documentation on creating my own builds.

powercycle, 2017-09-19:
As far as development goes, does this get you going in the right direction? https://github.com/opentx/opentx/wiki

teckel12, 2017-09-26:
I found the nightly builds for the Companion, but not the firmware.  Compiling hasn't been easy, and I'm trying to get the free memory on my Q X7 so I need a pre-release to diagnose a problem with iNav telemetry.

teckel12, 2017-09-27:
All sorted out as far as nightly builds. Will investigate issue now.

teckel12, 2018-05-19:
All good in 2.2.1

closed 2018-05-19 by teckel12