opentx/opentx #5226: Recent files and Profiles menu broken in menu bar on OS X

schwabe, 2017-09-19:
On OS X the recent files and profiles menu in the menu bar are greyed out/disabled. The function work in the toolbar.

Commenting out this in Mainwindow.cpp fixes the problem but make the toolbar menus not so nice

```
  // workaround for default split button appearance of action with menu  :-/
  /*QToolButton * btn;
  if ((btn = qobject_cast<QToolButton *>(fileToolBar->widgetForAction(recentFilesAct))))
    btn->setPopupMode(QToolButton::InstantPopup);
  if ((btn = qobject_cast<QToolButton *>(fileToolBar->widgetForAction(profilesMenuAct))))
    btn->setPopupMode(QToolButton::InstantPopup); */
```

mpaperno, 2017-11-29:
Hmm, I'm not seeing this on Sierra... everything in menu and toolbar works.  Qt 5.7.1, clang 800.0.42.1, OS 10.12.2 (looks like I need to update).  Especially strange since the code you quoted above should specifically affect only the toolbar button, not the actual action which it represents (or its appearance in the menu bar).

schwabe, 2017-11-29:
Hm,  Qt Version: 5.9.3 here. Now the problem is gone, might be a problem in 5.8. Closing this for now

closed 2017-11-29 by schwabe