opentx/opentx #5229: Errors on write model to radio

smokeintheeye, 2017-09-21:
hello all, im getting a bit stuck on writing my models back on to my taranis qx7. i have a screen shot on what open tx comes back to me with but im lost on how to fix it. 
[image]
[image]

3djc, 2017-09-21:
This is not a support site, but a bug tracking system. Please seek help in forum like rcgroups or openrc, they are better place to seek usage help

smokeintheeye, 2017-09-21:
Ok no bother just i was thinking it might have bean a bug with it happening in the nightly build. can read all the models off the radio but just cant seen to write out back on.

3djc, 2017-09-21:
You mean those model s where read from x7 ?

smokeintheeye, 2017-09-21:
yes i can read all setting and models off the radio but if i got to edit any of them and then go to write the models back to the x7 i get the errors show above. just a bit lost on how to fix it. looking it up now but cant seem to fine out that helps yet. iv run as admin backed everything up reformatted the mem card still the same. any help will be amazing. thanks

3djc, 2017-09-21:
Then could you please post your otx ? Looked liked it was coming from a different radio type

smokeintheeye, 2017-09-21:
the open tx im using is companion-windows-2.2.1N384. same for firmware and x7 bootloader -2.2.1n384. will go back to stock 2.2 see if that fixs it. but i like the new 2.2.1 the holding the power on the turn on is loads better. shutdown logo is a nice touch as well.
[image]
[image]

schwabe, 2017-09-21:
We need the actual otx file to debug this.

smokeintheeye, 2017-09-21:
ar right you will have to point me in the right direction to what one you want and its location. i can send you the models or eeprom or back up. but im not 100% on the right OTX file to send

kilrah, 2017-09-21:
Use "Backup radio to file" and send the resulting file.
Looks like the data comes from another radio type, and the conversion needs manual intervention.

smokeintheeye, 2017-09-21:
hope this helps. after updating to the nightly build i reset the full radio to stock reformatted to sd card and copied the vr11 sd content over. all models have bean made new on the x7 its self. thanks for all your help on this.
[x7 Backup.zip](https://github.com/opentx/opentx/files/1322745/x7.Backup.zip)

smokeintheeye, 2017-09-22:
just updated to the new build of opx still getting the same error? any news on what i can do to fix this?

kilrah, 2017-09-22:
Something apparently got broken. Use companion 2.2.0 instead of nightlies.

schwabe, 2017-09-23:
[image]
[image]
Something gone terrible wrong here

schwabe, 2017-09-23:
Git bisect tells me that I broke this in 87295776ff923314e0de699bc151d2ee3deb9360. But I am not seeing any change that affects the X7.

closed 2017-09-23 by schwabe

smokeintheeye, 2017-09-24:
I still get the same errors even after updating to the last 2 nightlys.is it something I'm doing wrong or is it a bug in open tx? Thanks for all your help on this.

kilrah, 2017-09-24:
Please read what we told you... It was a bug, and @schwabe fixed it yesterday, so the next nightly should be OK.

commit a502a997 (2017-09-23): Fix logic mistake I made earlier (closes #5229)