opentx/opentx #5230: Update for QX7S

reptil1990, 2017-09-22:
I got the new QX7S and wanted to flash a different configuration. When is support coming or did I miss something?
Thank you!

3djc, 2017-09-22:
Support of X7S will be coming in 2.2.1

closed 2017-09-22 by 3djc

kilrah, 2017-09-22:
Just flash X7... they're the same.

reptil1990, 2017-09-22:
> Just flash X7... they're the same at this point.

I am a little bit concerned because the QX7 has no trainer function

reptil1990, 2017-09-22:
@3djc what is the timeline for the 2.2.1 release? Would be happy to test and give feedback!

kilrah, 2017-09-22:
There is no timeline. 

X7 firmware has trainer function, just not over bluetooth. 
Bluetooth trainer is currently not a OpenTX feature on any platform anyway. It will be added soonish, but not sure it will make it in 2.2.1. 

You can test the nightly builds if you wish.

schwabe, 2017-09-22:
@reptil1990 btw. what build information does your released firmware on your X7s has?

reptil1990, 2017-09-22:
i can send you the dump i made before i flashed the current 2.2

reptil1990, 2017-09-22:
[origFirmarQX7S.bin.zip](https://github.com/opentx/opentx/files/1325582/origFirmarQX7S.bin.zip)

that was on the radio