opentx/opentx #5232: No Radio Mapping for Taranis Q X7S

SuperSurferSaurus, 2017-09-23:
Hi All, I just got a Taranis Q X7**S** (there seems to be some difference with the S mode)
For some reason the companion wont let me connect, the radio isnt a variant to select.
There may be some additional channels - or a variant of switches - any help finding a way to connect would be awesome.

Regards
Ev

schwabe, 2017-09-23:
Just select X7. The hw of both is identical apart from bt trainer that is not yet supported by OpenTX anyway.

closed 2017-09-23 by schwabe

SuperSurferSaurus, 2017-09-23:
Hi All,
Thanks Arne, yeah I tried that innitially,
Currently the radio has a customised version of X7 Bootloader 2.2.0 (Has
alpha-numeric version?)
I am trying to replace that with 2.2.0 (85863358) with the file
opentx-x7-noheli-lua-en.bin

But when I try to write to radio - I get this error.
New Firmware is not compatible with one currently installed.

Thanks

SuperSurferSaurus, 2017-09-23:
ah, turn hardware compatibility test off, sorry my bad,
Thanks