opentx/opentx #5234: Feature request: Bluetooth Joystick

reptil1990, 2017-09-23:
Do you think it would be possible to implement a Bluetooth Joystick emulation for Flight Sims?
So you can bind your Taranis wit BT (QX7S) to your computer and use it wirelessly.
I don't know how the latency would be... 
What do you think?

schwabe, 2017-09-23:
Short version: Theoretically yes. But pratically no. 

The firmware on the Bluetooth modules only supports serial link. For a bluetooth Joystick emulation you would need a completely different firmware on these modules.

closed 2017-09-23 by schwabe

kilrah, 2017-09-23:
And as far as I know there is no existing standard for Bluetooth Joystick in the first place, so that would be a PC coding project outside of the scope of OpenTX.

reptil1990, 2017-09-23:
I looked around and found some software to convert serial to HID. We just need a way to bind it to a PC. Is it possible to implement?

garyjohnson, 2017-12-12:
Recently got a QX7S and implemented this for macOS: https://github.com/garyjohnson/opentx-ble-joystick. Once I feel that is in a polished state I hope to look into writing a Windows version.

reptil1990, 2017-12-12:
Amazing I will have a look asap!!!

Riffer, 2020-04-23:
@garyjohnson from my understanding, this would be useful to connect **any** OpenTX device (that includes BT like the T16 ie.) to any gaming device (Smartphone with Freerider?). Did you progress in your afford to make it usable the way? Or does this always need a HID-plugin on the other side?