opentx/opentx #5235: Companion 2.2.1 Higher Input Sources Disappear

r41065, 2017-09-25:
[image]
[image]

First Time Import Models from OTX 2.2 I wonder that Simulator didnt start
Using Higher Input Numbers all are shown with ??? as Source.
Write all Source manual in again and save them but next time i start this Model the sources are dissappeared again

looks that only the first 4 Inputs-Lines  are saved to OTX-File

Radio Type is Taranis X9d

[Jan221.otx.zip](https://github.com/opentx/opentx/files/1328746/Jan221.otx.zip)

Ralf

r41065, 2017-09-25:
Operating System from Computer is Win 7/8

kilrah, 2017-09-25:
What nightly? Probably fixed on today's one.

r41065, 2017-09-25:
N380..N387  the other i have not tested

kilrah, 2017-09-25:
Yes looks OK. Always check on the very latest before reporting an issue.

[image]

closed 2017-09-25 by kilrah

r41065, 2017-09-25:
Sorry but since Yesterday i get only a network Time Out Error if i try to connect 
to http://downloads-22.open-tx.org/nightlies/companion/windows/

N387 was the last version i got...

kilrah, 2017-09-25:
Ah yep, server went down, and also didn't build a nightly as it broke before the scheduled time... you're forgiven ;)

kilrah, 2017-09-25:
Today's nightly is now up if you want to test.

r41065, 2017-09-25:
N388 looks like it works correct:
Thank You
Ralf