opentx/opentx #5236: Update wiki module bay pin out

RCdiy, 2017-09-25:
https://github.com/opentx/opentx/wiki/Taranis-serial-port#taranis-module-port-pinout

Has

name | function | PC connection
-- | -- | --
PPM | PPM/PXX/DSM output |  
HEARTBEAT | SBUS/CPPM input |  
BATT | switched battery voltage |  
GND | ground |  
SPORT | telemetry output

Maybe update the to

Name | Function | 
-- | -- | 
OUT | PPM/PXX/DSM2/DSMX/SBUS/Serial |  
IN | SBUS/CPPM |  
BATT | on/off direct battery voltage |  
GND | ground |  
SPORT | telemetry output

yanivasy, 2017-09-28:
As far as I know, SPORT was used for also CrossFire which is bi-directional and not only an output.
The switch battery voltage seems self-explainarry.
The HEARTBEAT name is used thru-out the code so changing it might lead to confusion.
As for the PPM name changing to out- might be right.
Currently, the PPM out doesn't support SBUS, please point me to its serial ability.

RCdiy, 2017-09-28:
Just implemented.... sbus in a nightly... that's how I found out and wondered if wiki could do with updating...

https://opentx.gitbooks.io/manual-for-opentx-2-2/content/model_setup.html
rcdiy 2:03 PM
Manual, link above - under external module says "SBUS to output SBUS". I check on the simulator and on my radio, SBUS not showing up...

Kilrah Admin Owner 2:13 PM   
was added a couple of days ago. Not in release yet.

yanivasy, 2017-09-29:
The SBUS output implementation was started several years ago (see issue #2894 : https://github.com/opentx/opentx/issues/2894) and is currently in the pull requests (#5225  : https://github.com/opentx/opentx/pull/5225) but not as a milestone for the coming OpenTx 2.1.1.

As for serial output stated for PPM output pin (other format rather than SBUS): please point me out to any.

geofrancis, 2017-09-29:
I know a lot of people were wanting this feature to use with LRS systems, openlrsng had sbus in and out.

3djc, 2017-09-29:
Personnaly, I do not mind the updated description (albeit it is difficult to keep up to date) but I would not change pin names, as many existing tutorial, docs, ... are refering to those names

RCdiy, 2017-09-29:
the sbus made its way into the gitbook manual... so I figured it is was put in there then what about the wiki and figured lets put a place holder so an admin may get to it...

kilrah, 2017-09-29:
> As for serial output stated for PPM output pin (other format rather than SBUS): please point me out to any.

DSM which has been there forever and MULTI (which is basically SBUS with 2 extra bytes to pass configuration data).

schwabe, 2017-09-29:
@yanivasy Multi output is probably the best thing for you to use if you have some TX module unless you *really* require SBUS for something that only speaks SBUS. Multi has the advantage of having RX number, and a few other options that can be selected. And with Multi we can implement further things if you need additional inputs.

schwabe, 2017-09-29:
@geofrancis orls already has multi input, which is better suited for this implemented already.

yanivasy, 2017-09-29:
> DSM which has been there forever (ah, mentioned already) and MULTI (which is basically SBUS with 2 extra bytes to pass configuration data).

schwabe, 2017-09-29:
@yanivasy To repeat myself. There is already a pull request for exactly implementing SBUS. It uses VBAT voltage but since it is current limited by a 10k resistor directly connecting to a FC is still fine.

RCdiy, 2017-10-16:
Could an admin update the wiki?

kilrah, 2017-10-29:
Done

closed 2017-10-29 by kilrah