opentx/opentx #5237: Taranis Q X7: no Haptic in Only Alarms mode

janbolt, 2017-09-25:
In Haptic Mode "Only Alarms" Haptic does not work,
only in Mode "All".

Taranis Q X7
FW 2,2,0

twistedwings, 2019-08-17:
Rather than marking the issue as stale wouldn't it be better to fix it?
It's still there in 2.2.4.

raphaelcoeffic, 2019-08-17:
@twistedwings the issue are now marked "stale" automatically, just like many other projects do. By replying you show your interest in a particular bug/issue, so that we know it is still a bug, or still an interesting feature. ATM, we are just swamped in zillions of issues that nobody ever touches again...

So bottom line is: it's very good that you care about that issue and write that it is still a bug ;-)