opentx/opentx #5239: Crossfire protocol implmented from OpenTX 2.2

krzysztofkuczek, 2017-09-26:
Hi,
Can you share any documentation regarding crsf protocol you are implementing from version 2.2.
I took a loot at source code, and it's quite different to the implementation i found in betaflight... event bitrate is different  420000 vs 400000 ?

I'm going to force my lrs to work with taranis ;)
Thank you in advance
Krzysiek

3djc, 2017-09-26:
No sure what you are asking:  our code is open to have a look at, and crsf is a product of TBS, they are the ones you should be asking for specs and documentation if I understood you correctly

krzysztofkuczek, 2017-09-26:
Ok,
I just thought that, there is free available crsf protocol documentation...

schwabe, 2017-09-26:
@krzysztofkuczek for olrs there is already an implementation of the multi protocol that OpenTX also supports. 

And also it might be entirely possible that rxs use a different buad rate and protocol than the tx

krzysztofkuczek, 2017-09-26:
Thx, 
But there seems to be no mult protocol documentation neither ;)
Except source code :) or maybe i can't find such a basic information like frame format, bound rate on it's doc page https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/docs

Regards
Krzysiek

schwabe, 2017-09-26:
OpenTX has a pointer to the docu and DIY has the douc in the header

https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/blob/master/Multiprotocol/Multiprotocol.h#L437

If you have additional questions etc. feel free to ask or drop by in our chat. There is also a pull request open from that implements sbus output if you looking for a simple standard protocol.

krzysztofkuczek, 2017-09-26:
Schwabe, Thank you veru much :)

closed 2017-10-29 by kilrah