opentx/opentx #5240: LUA lcd.drawBitmap on X7 doesn't work

NiklasVoigt, 2017-09-26:
Hi, i've checked out commit 541aa459cf5e368dbff916553e4124abc43e2fb9 for testing drawBitmap for the X7, but this doesn't work.

LUA:
lcd.drawBitmap(0, 0, "plane.bmp")

CLI:
Script error: wizard.lua:40: attempt to call field 'drawBitmap' (a nil value)

Thanks!

3djc, 2017-09-26:
It does

```
-- Main
local function run(event)
  lcd.clear()
  lcd.drawPixmap(1, 1, "plane.bmp")
  return 0
end

return { init=init, run=run }
```

[image]

closed 2017-09-26 by 3djc

schwabe, 2017-09-26:
you need a nightly build. 2.2.0 does not have the feature

NiklasVoigt, 2017-10-03:
now i am on 2.2.1 but the image does not displayed..
the graphic i used is an 1bit bmp file, maybe you can add yours..

[image]

Lua
```
-- Main
local function run(event)
  lcd.clear()
  lcd.drawPixmap(1, 1, "/plane.bmp")
  return 0
end

return { init=init, run=run }
```

CLI
```
findTrueFileName(/home/niklas/Desktop/sdcard-taranis-x7-2.2V0011/plane.bmp)
	found in map: /home/niklas/Desktop/sdcard-taranis-x7-2.2V0011/plane.bmp
f_open(/home/niklas/Desktop/sdcard-taranis-x7-2.2V0011/plane.bmp, 1) = 0x7f4fd4002270 (FIL 0x7f4ffccafdc0)
f_size(0x7f4fd4002270) 544
f_close(0x7f4fd4002270) (FIL:0x7f4ffccafdc0)
```

3djc, 2017-10-03:
I tested with one from here: http://www.open-tx.org/screens-9x.html

NiklasVoigt, 2017-10-03:
same behaviour in CLI with a png file from the link

3djc, 2017-10-03:
```
convertToSimuPath(): plane.bmp -> plane.bmp
findTrueFileName(plane.bmp)
	found in map: plane.bmp
f_open(plane.bmp, 1) = 03EEEAC8 (FIL 085BF024)
f_size(03EEEAC8) 318
f_close(03EEEAC8) (FIL:085BF024)
```

3djc, 2017-10-03:
[plane.zip](https://github.com/opentx/opentx/files/1352472/plane.zip)

3djc, 2017-10-03:
I guess you did not see the footprint on max pict size

NiklasVoigt, 2017-10-03:
Thank you, now it works! How much is the max size

3djc, 2017-10-03:
Maybe it would be wise to look for that info on opentx lua manual ?

floaledm, 2017-12-21:
Have any of you tried drawPixmap() on the X7 with images narrower than 25 pixels? I can't seem to get any image of 24 pixels wide or less to draw correctly... No problem on the height (as long as it is 64 pixels or less).
Here's the code, the files I used, and a screenshot of the result. test1.bmp is 25 pixels wide. test2.bmp is the same image cropped by one pixel on the right, so it is 24 pixels wide.
```
lcd.drawPixmap(1, 1, "/test1.bmp")
lcd.drawPixmap(30, 1, "/test2.bmp")
```

[test.zip](https://github.com/opentx/opentx/files/1577519/test.zip)

[image]

Pardon my ignorance, but what is the max image size on the X7? 64x64?