opentx/opentx #5248: OpenTX 2.2.0 - AR9x board - Companion 2.2 - can't save models

DaveKraft, 2017-09-28:
It worked great for a while, but then (and I don't know why) it suddenly stopped, now I can read out my models from my transmitter, but when I try to write them back I get two messages:
1.  "Cannot write model ......Source ??? Cannot be exported on this board!"
and when I close this, the second appears
2. "Error - Cannot write temporary file"
Any suggestions what is wrong?

Thx
David

kilrah, 2017-09-28:
Sounds like you may be using a nightly build of companion istead of 2.2.0 release.

DaveKraft, 2017-09-28:
No, I use the current version(from date 30.5.2017), I even uninstall completely my companion 2.2, download actual version from opentx site and install again, and it is still the same.

kilrah, 2017-09-28:
Can you supply your backup file?

DaveKraft, 2017-09-28:
yes, I can. Here is my radio backup.
[fullbackup-28.9.2017.bin.zip](https://github.com/opentx/opentx/files/1340464/fullbackup-28.9.2017.bin.zip)

schwabe, 2017-09-28:
I think the AR9X might be affected by the bug I fixed for that platform (#4961). Can you download the latest daily build and check if that is fine? Looks good to me with my current build.

DaveKraft, 2017-09-28:
sure I can, but to be honest, I'm not entirely sure what is last daily build and where I can find it? :-) can you post link, or something?

kilrah, 2017-09-28:
Opening this file with current next and clicking simulate gives the errors, so either it's not fixed or the file is already in a corrupted state.

closed 2017-11-08 by kilrah