opentx/opentx #5250: DSM (DSM2, DSMX, LP45) output on Taranis X9D+ connected to Pixhawx (PX4) SPKT/DSM input

yanivasy, 2017-09-29:
I've been trying to connect directly (hard wire) Taranis X9D+ to a Pixhawk flight controller (PX4).
I connected the PPM output pin within the Taranis external module thru a resistor based voltage divider (two 10kOhm resistors) to the Pixhawk spectrum/DSM input.
A voltage divider was needed due to the 7V level of the PPM output which is behind the 3.3V input specification of the SPKT/DSM input.

Using the original FrSky firmware (I think 2.1.6) that enable 1-8 CH:
When the Taranis X9D+ was configured to DSMX  the Pixhawk flight controller detected the RC but readings **where not stable**. This was the case also with DSM2 and LP45. 

Using the original OpenTx 2.2.0 firmware that enable only 6 channels (default CH 1-6) :
Pixhawk flight controller **couldn't detected** the RC at all.

When changing the channel to 3-8 (I think), once again, the Pixhawk flight controller detected the RC but readings **where not stable**.

From what I figured out by now, Taranis+ PPM output (set to DSM mode) uses 125,000 bps baud rate, whereas the Pixhawk flight controller DSM input was programmed to 115,200 bps baud rate ( https://pixhawk.org/dev/radio_control?s[]=rc&s[]=ppm&s[]=dsm ).

1. Are the baud rates as mention above?
2. If so, do these differences might cause the unstable RC readings?
3. What causes the OpenTx 2.2.0 firmware update to the channel reduction (I saw #5204)?
4. Why does the new firmware affect the Pixhawk  SPKT/DSM input?

schwabe, 2017-09-29:
I think it is all explained in #5204. 2.2 changed to 125000 8N1  instead 125000 8N2 since that is what real Spektrums uses. Also channel numbers has always been 6ch (apart from the display bug).

125k vs 115,2k baud rate is asking for problem.

But DSM2 as serial input is probably the worst problem to choose from as input. Iam not even sure why flight aware MULTI or SBUS (when merged) are probably much better as input. I am not even sure in what scenario a FC needs DSM2 input.

yanivasy, 2017-09-29:
I agree that Spectrum's DSM is 125,000 8N1 (https://www.spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf) and that the 2.2.0 change was in place.
I can't figure out how DSM worked before this correction and how the Pixhawk detects DSM with the bug and also detects DSM after changing the channels without the bug (in 2.2.0).
Spectrum declare for long range capabilities- might be the reason for Pixhawk hardware support
.    
I'll give the Multi a try. It a 115,200 8N1, right?

schwabe, 2017-09-29:
no like in the multi protocol specification 100k 8e2 (basically like sbus but not inverted). Also you can compile/get a compiled sbus pull request if you want. Watch out the channel channel on multi is different (more logical) compared to SBUS.

yanivasy, 2017-09-29:
A compiled SBUS firmware will help a lot since I'm kind of rusty with C++ (currently programming the good old fashion assembly). Thanks.

schwabe, 2017-09-29:
@yanivasy build options and platform?

kilrah, 2017-09-29:
> I am not even sure in what scenario a FC needs DSM2 input.

Use of Spektrum satellites, but I believe the protocol for that is not identical as what the modules expect (and we thus send) so that's likely why it doesn't work.

yanivasy, 2017-09-30:
> Use of Spektrum satellites as receivers, but I believe the protocol for that is not identical as what the modules expect (and we thus send) so that's likely why it doesn't work with a direct connection.

@kilrah, according to Spectrum's DSM specification (see a link above) "The packet is transmitted at 125000bps, 8 bits, No parity, 1 stop (8N1). For those UARTs which are not capable of that speed, they will also work at the more-standard 115200bps...".

Therefore,I quite sure now that OpenTx 2.2.0 is compatible with DSMs protocols.
Probably the Pixhawk flight controller DSM input doesn't comply with Spectrum specifications of baud rate although a 115,200 bps baud rate is allowed by Spectrum's specifications.   

> build options and platform?

@schwabe, my platform is Taranis+   X9D plus, as for build options:  Multimodule & sqt5font will do.

Thanks a lot :-)

schwabe, 2017-09-30:
The receiver protocol is different from the tx protocol implemented in OpenTx.

@yanivasy the pull request is merged, just use the latest nightly

yanivasy, 2017-09-30:
> The receiver protocol is different from the tx protocol implemented in OpenTx.

@schwabe , do you mean by "**receiver**" to the OpenTx RC's input (HeartBeat), to the flight controller input or to DSM transmitters modules ?

geofrancis, 2017-09-30:
if dsm can work at 115200 non inverted, could it be used over bluetooth?

schwabe, 2017-09-30:
The protocol that you liked is the    of an rx (to main rx or to fc). The dsm2 protocol that OpenTx implements is the protocol  a dx6i speaks to its internal sender module. They are **not** the same ir compatible

geofrancis, 2017-09-30:
thats unfortunate, as far as i know every other protocol uses 100k thats not really compatible with much.

schwabe, 2017-09-30:
@geofrancis we have no idea what you're redid l trying to achieve

geofrancis, 2017-09-30:
@yanivasy is trying to wire a taranis directly to a pixhawk using DSM2, if that was possible at 115200 non inverted then theoretically it wouldnt have to be just a wire between the taranis and pixhawk, you could put any transparent serial link between them like bluetooth or wifi and have a basic RC system.

i have looked at sbus over bluetooth before but the problem was the 100k baud it wouldnt work with any bluetooth modules.

yanivasy, 2017-09-30:
@geofrancis , according to @schwabe : the DSM protocol implemented over the Taranis PPM output (intended for an internal transmitter) is not the same as the one output by a receiver (intended to a FC).
I'll check it again tomorrow and try to verify.

I thought of making two baud rate adapters: one for 125,000->115,200 and the second for 115,200->125,000. In the future, the first one can be eliminated with some opentx adjustment for output 115,200 directly whereas the second one is unnecessary according to Pixhawk specifications. 
Same though cross my mind in respect to SBUS but before that I need to see that it's working by hard wire.
Have you tried the latest nightly including the SBUS?
If you have a build bin file for Taranis+ with the new SBUS ability, please share it since I'm in the progress of installing qt...

kilrah, 2017-09-30:
> thats unfortunate, SBUS protocol uses 100k thats not really compatible with much

SBUS is compatible with hundreds of devices, many more than DSM. 

> If you have a build bin file for Taranis+ with the new SBUS ability, please share it

Why would you need/want that since you can now get it yourself as nightly?

yanivasy, 2017-09-30:
> Why would you need/want that since you can now get it yourself as nightly?
I'm new to OpenTx and can't figure out how to get it as nightly - I'm currently installing Qt for building it.
If it already build, please point me out how to get it.
Thanks and sorry...

kilrah, 2017-09-30:
Link is on the opentx website.

yanivasy, 2017-09-30:
@kilrah , the nightly companion 2.2.1N392 downloaded from OpenTx WEB can't get any updates and therefore can't build a bin file. Is it just a problem in my PC?

schwabe, 2017-10-01:
We have sbus inverted and non inverted in nightly. I don't anything that already speaks sbus but not 100k. The bt link does not count, it is a stupid alternative to just using  an xsr or so

geofrancis, 2017-10-01:
Sbus uses 100k and that's not a standard baud rate so few serial devices will run at that speed. Only rc devices that are made for sbus run at that speed. I was thinking more generic serial radio hardware so anything can be used for a rc link.

XSR is 10x the price of a BT module

schwabe, 2017-10-01:
Yes we are quite aware of that. But is also TTL level and not serial. And as micro processors go, e.g. STM32 and AVR these are more happy with baudrates that evenly divide their clock frequency (e.g. 8Mhz), so 100k is just fine for them. When you go into embedded processors and circuits you will see these baudrates like these more often than not.

Also it is usually better to stick to established standard (e.g. SBUS) than to brew your own not quite standard solution.

a simple promini with serial(125k)+softserial(115200) might be enough to convert for your tests. 

RC world just uses these strange protocols. We are against your project but I am not seeing myself implementing something where I cannot really see the use case

geofrancis, 2017-10-01:
Fair enough, I was only interested if it was already possible I'm not looking for anything to be added.

yanivasy, 2017-10-01:
Maybe by using the serial port (within the battery bay) it will be easier and more code efficient to get a standard RS232 baud rate (https://github.com/opentx/opentx/wiki/Taranis-serial-port#taranis-module-port-pinout).
Using a second micro (such as the pro mini) just for translating one baud rate to another is good if you can't interfere with the code. However, OpenTx transmitters were made just for that: enabling veritable codes to be used by all types and vendors.
In either case, another baud rate translator will be required at the FC side- **but just one**.
**Latency** will also be reduce by eliminating adapters on path.

schwabe, 2017-10-01:
@yanivasy serial port used on the X9D has nothing/little to do with the baud raute. At the moment there just no compelling reason to do so. 

SBUS is 100k, DSM TX internal is 125k, MULTI is 100k. We implement protocols not baud rates. So all protocols implemented so far just happend to have these baud rates. 

Latency is negliable if you just read/write a character at a time(1/10 ms for 100k baud)

I personally don't think implementing SBUS with a different baud rate is a  good idea at all. 

As you point out that is OSS. However, that also means people will only do/implement what they think is right.

I think what you really want is a MAVLINK input/output implementation on the X9D. It also is defined at other baud rates. If you really want to help think about contributing a MAVLINK implementation.

yanivasy, 2017-10-02:
**MAVLINK implementation sounds like a fantastic idea.**
Such a suggestion is not a minor change/modification so it will require a great deal of work. 
Currently I'm using beside the Tarins, a laptop with a set of transceivers to control, program and get telemetry data -  I think it a certain type of MAVLINK. Will the Taranis's display be enough to put all information or maybe splitting it to several pages can do the trick. 
From the hardware perspective your idea requires both uplink and downlink communication ability of the Tx. I'm not familiar with the XJT capabilities: are they suitable for it?
If not, an external transceiver module will be needed: can be connected by using either the serial port connector or the external Rf module connector (since most transceivers use in and out lines, the HeartBeat can be used as input whereas the PPM/SPort as output line).
It autopilot mode, the use of Taranis is mainly as backup/emergency. I think that this ability should be kept-in the for-coming MAVLINK edition

schwabe, 2017-10-02:
Complete mavlink implementation will probably never happen. Mavlink needs a serial link, the XJT is black box from OpenTX's perspective and will only Frsky's radio protocol. But you could just start with decoding basic telemetry and/or sending out channels information.

Heartbeat and Sport are both inverted uarts, so a module has to deal with that.

yanivasy, 2017-10-02:
> Heartbeat and Sport are both inverted uarts, so a module has to deal with that.

@schwabe , Just as you implemented SBUS ("normal" or "not inverted"), it can be dealt with for any standard transceiver.

BTW, the SBUS seems to work all right (therefore I didn't tried the MULTI outputs) - thanks.
As for the DSM protocol, you are right that it's different within the Pixhawk  compered to the OpenTx implementation for the hacked modules.

schwabe, 2017-10-02:
@yanivasy the sbus and heartbin ins have hw inverters so receiving anything non inverted is problematic on them.