opentx/opentx #5257: Taranis X9D simulator issues on mac

mcvd02, 2017-10-01:
Updated radio to 2.2.0 and now my radio does not work on simulator. Mac recognizes radio but not sensing any of the inputs.Radio worked perfectly prior to updating.

schwabe, 2017-10-01:
I cannot reproduce this. Did you check in setting if your remote is displayed there?

mcvd02, 2017-10-01:
On my computer? If thats what your asking yes it is displayed there.

schwabe, 2017-10-01:
Is it displayed in finder? Can you make a screen of the system information screen with the Taranis selected? Like this

[image]

mcvd02, 2017-10-01:
Yes mine shows up

mcvd02, 2017-10-01:
[image]

schwabe, 2017-10-01:
Also selected as the joystick in Companion's preferences? (Btw. you make a screen with CMD+Shift+4)

mcvd02, 2017-10-01:
Can you send screenshot? Only setting I thought you had to uncheck was the mass storage option

schwabe, 2017-10-01:
[image]

But for the future is rather a bug tracker than a help support forum

mcvd02, 2017-10-01:
This is potentially a bug and it is enabled. Like I said had no issue prior to updating.

kilrah, 2017-10-01:
Did you also update your mac? Joystick on mac is messy. Never managed to have it work without an utility here, don't have the name at hand.

schwabe, 2017-10-01:
also please provide as much information as possible. It is annoying to ask you every little detail. Do the sticks work in the calibration box on the companion screen? Does the joystick feature work in other apps, etc.

kilrah, 2017-10-01:
Like when you say "simulator" @schwabe takes it to be companion's radio simulator. I suspect you might be talking of a third party flight simulator instead, but since you didn't give details we have no idea.

mcvd02, 2017-10-01:
Update-Mac recognized FrSky Taranis joystick but companion does not. Say joystick not found in companion simulator settings.

schwabe, 2017-10-01:
@mcvd02 You are contradicting yourself. In https://github.com/opentx/opentx/issues/5257#issuecomment-333383512 I asked you if your Taranis is recognized there in Companion and you said you have it enabled. And now you say that the Joystick is not found by Companion? 

Sorry, what you are writing so far is to imprecise and vague to get a picture what is really happing on your side and what you are actually doing.

mcvd02, 2017-10-01:
It was enabled before I updated in companion. I just plugged in controller to double check and it is not being recognized in companion. This is the current condition. My mac recognizes and companion does not. Sorry for the confusion.

mcvd02, 2017-10-01:
3rd party flight simulators recognize joystick but none of the inputs/channels working.

schwabe, 2017-10-01:
Taranis might need to connected before starting companion. Not 100% if you update that list

mcvd02, 2017-10-01:
Connected Taranis prior to opening companion, still does not recognize.

mcvd02, 2017-10-01:
Any recommendations??? Why would stock Taranis X9D Plus SE work stock and after updating not work?

schwabe, 2017-10-01:
All I know is that it works fine with macOS sierra and High Sierra. And your bug report has not given enough information to even try to reproduce your problem, sorry!

mcvd02, 2017-10-01:
OS X Yosemite 
version 10.10.5
What other information do you need?

mcvd02, 2017-10-01:
Companion recognizes taranis joystick now but going through calibration, none of the channels are being recognized. I was in boot loader mode earlier trying to get simulator on companion to recognize it.

closed 2017-10-01 by mcvd02

powercycle, 2017-10-02:
To be clear, for the mac (and I think for the others as well) you just plug it in to the computer and it recognizes the transmitter as a joystick, right?

kilrah, 2017-10-02:
You do need to set up appropriate mixes on the radio though of course.

powercycle, 2017-10-02:
Would not having mixes explain the output issue that @mcvd02 is experiencing?

mcvd02, 2017-10-02:
Can you point me in the right direction to set these up? Like I said it was working perfect prior to update. Thanks

mcvd02, 2017-10-02:
I have set weight and offset to 50, still nothing

schwabe, 2017-10-02:
Can you try to test your Taranis on a different PC/Mac to check what part of your system broke? Or backup your eeprom and flash back to the old version and see if it changes anything?

schwabe, 2017-10-02:
The Taranis HID just implements a standard HID joystick. There hasn't been much changes from 2.1 to 2.2 (other than 24 instead of 8 buttons)