opentx/opentx #5260: PXX (XJT) FLAG1 seto to PXX_SEND_FAILSAFE randomly (every 800-1000 frames)

krzysztofkuczek, 2017-10-02:
Hi,
I just found strange issue, it looks like OpenTX 2.2 (running on Taranis Plus) sends randomly fail safe values, when nothing like this should be send.
Let see log for XJT D8 mode log grabbed directly from Taranis external module port...

126,0,64,0,0,228,63,0,1,64,0,1,112,0,193,111,0,117,126,
126,0,64,0,0,228,63,0,1,64,0,1,112,0,193,111,0,117,126,
**126,0,80,0,255,247,127,255,247,127,255,247,127,255,247,127,0,56,126,**
126,0,64,0,0,212,63,0,1,64,0,1,112,0,193,111,0,228,126,

80($50) instead of 64($40), it happens quite randomly, i just calculated difference between occurrence (see attached dump) and it's like following

frame no difference
875.5
873
1123
625.5
874.5
875.5
3498

So in most cases, every 800-1000 frames, fail safe value is send... I don't know what values are send ;) 

You can see raw data ...
[pxx_dump.zip](https://github.com/opentx/opentx/files/1349206/pxx_dump.zip)

Regards
Krzysiek

schwabe, 2017-10-02:
well yes failsafes value do not need to be send in D8 mode but it also does not hurt to send them, the module will just ignore them. So we just use PXX the same way for D8/D16/LR12 and R9M

closed 2017-10-02 by schwabe

krzysztofkuczek, 2017-10-02:
Yes, but same problem is in D16 mode. Maybe i don't understood it but FS values should be send only on user request...
Krzysiek

schwabe, 2017-10-02:
PXX specifies them to be send periodically. The TX module does not save the failsafe values but just forwards them to the RX (in D16 mode).

krzysztofkuczek, 2017-10-02:
Ok. But what valued you send, if setting fail save has not been never requested by user...? What will happen if Rx will use those values as fs values...

schwabe, 2017-10-02:
For D8 the values are ignored anyway. And for D16 there is a flag that is "no failsafe set"

krzysztofkuczek, 2017-10-02:
Well, in my opinion, if D8 does not support failsafe setting from radio menu, it means that D8 compatible module, does not need to recognize  PXX_SEND_FAILSAFE flag. It means, that it will process frame with fail safe flag set as normal channels values which may cause real problems.

For D16 do you mean this piece of coding

`if (g_model.moduleData[port].failsafeMode != FAILSAFE_NOT_SET && g_model.moduleData[port].failsafeMode != FAILSAFE_RECEIVER) {
`

Krzysiek

schwabe, 2017-10-02:
There are no d8 only modules that we support with pxx. Only pxx modules we support are xjt and r9m

schwabe, 2017-10-02:
I agree that sending failsafe frames for d8 is probably mot needed but since this code has been that way for years now, I would rather keep it as is

kilrah, 2017-10-02:
The XJT always handles failsafe data correctly regardless of mode. 

Failsafe being sent regularly is required as it is not possible to know (reliably) when the receiver is powered up or down.

krzysztofkuczek, 2017-10-02:
ok, thx for explanation.
It means, that fail safe values are related to model settings, and every time you choose different model, different fs values are transmuted to RX and stored in it's memory for use in case of loosing RF link...

kilrah, 2017-10-03:
Yes of course.