opentx/opentx #5261: Suggestion for saving memory with Lua

jfrickmann, 2017-10-03:
First of all, thank you for all of the work that you put into the OpenTX software. It is a truly amazing system!!

This is a feature request rather than an issue. I have a setup with three telemetry screens, and I have managed to barely squeeze it into Taranis' memory. It crashes the first time I load fresh files, but with the "luac" option it gets them compiled, and after rebooting it works. Now I would like to add a Lua function script to the setup, but here I am SOL, as the third telemetry script is now in a permanent state of memory panic.

I have played a little with "L Shems" Guido's solution to load and unload scripts, but as far as I can tell from probing used memory with collectgarbage("count"), it does not completely clean up. But I will give his solution a try.

In the meantime, I have the following more general suggestions for a solution that could help save memory with Lua telemetry scripts:

1. Add an "Exclusive" checkbox. This will run the script just like a one-time script is run today.

2. Add an "Unload in background" checkbox. When the script is not active, then it is unloaded from memory. E.g. in my setup, it is not necessary to have the scripts for browsing saved scores and for graphing telemetry data loaded in memory when they are not in the foreground.

Option 1 means that other Lua scripts are halted when that telemetry screen is selected. As long as we follow the advice of not having anything critical rely on Lua, this should not be a major problem. The script could be kept running when the telemetry screen is left by long pressing EXIT, and only stopped when another telemetry screen is selected.

Option 2 keeps everything else running, and is thus only effectively reducing the max memory usage if it is selected for at least two scripts.

Options 1 and 2 together would unload the script and start up everything non-exclusive, if the telemetry screen is left by long pressing EXIT.

Thanks again!
Jesper

jfrickmann, 2017-10-04:
I just discovered a stupid bug in my test code of Guido's solution. This means that it actually cleans up memory nicely. So this feature request is not as urgent anymore. But I still like my suggestions, so I will leave them here for you. Thanks!

lshems, 2017-10-04:
Actually, there is some mayor improvement in running Lua for the x7. Perhaps less memory needed for the screen, as it is smaller.

If it would not be that, it would be interesting to check.

Another unexplainable issue is that after unloading code, and bring the code in the exact same state, sometimes the same next action will crash, and sometimes not.

Still some issues so with Lua, but thanks for what has been achieved already.

The load unload scenario works well in most cases.

What I would like is have the go.lua script being launched as a one time script on model activation. And an option to start the model in a telemetry screen by default.

3djc, 2017-10-04:
Yes, x7 a more mem because for lcd size and it's only 1bit depth

jfrickmann, 2017-10-05:
I made a Lua script that automatically loads a script when its telemetry menu is selected, and unloads it when another telemetry menu with a similar autoloader is selected. It seems to work, so it is an alternative to making modifications to OpenTX.
[Autoloader.zip](https://github.com/opentx/opentx/files/1357761/Autoloader.zip)

Jesper

closed 2017-10-05 by jfrickmann