opentx/opentx #5262: companion: no D8 mode

janbolt, 2017-10-03:
I cannot select D8 mode in companion, in the radio I can.

Taranis Q X7
opentx 2.2.0

[image]

[image]

schwabe, 2017-10-03:
probably your companion is to eu. The simulator is always non EU, your radio might also have a non EU firmware

closed 2017-10-03 by schwabe

janbolt, 2017-10-03:
"probably your companion is to eu"

What does that mean ?
Is there a eu and non eu companion ?

schwabe, 2017-10-03:
options of your radio in settings.

janbolt, 2017-10-03:
there is no eu option in the radio settings.
Firmware is non eu.

schwabe, 2017-10-03:
There sure is one. See screenshot
[image]

janbolt, 2017-10-03:
I was assuming that as a firmware option only.
Anyway the "eu" is not checked in companion.

schwabe, 2017-10-03:
The eu option is the only things that controls if D8 shows up or not.

janbolt, 2017-10-03:
and why does it not show up ?

schwabe, 2017-10-03:
please double check that you have not selected the eu option.

janbolt, 2017-10-03:
eu is unchecked believe me

schwabe, 2017-10-03:
Try close/reopen companion. I have nothing I can anymore on my end. I cannot reproduce it and I looked through code and the code really only checks if eu is selected.

janbolt, 2017-10-03:
restart does not change anything.
The problem occurs on my notebook only.
On my desktop I cannot reproduce it either.

kilrah, 2017-10-03:
Try creating a new radio profile.

janbolt, 2017-10-04:
creating a new profile solved the problem