opentx/opentx #5263: Companion 2.2.1N395 Simulator Crashes at Telemetry Display

r41065, 2017-10-03:
[image]

Win 7, Companion2.2.1N395
[CompanionDebug_17-10-03_17-58-52.log](https://github.com/opentx/opentx/files/1353020/CompanionDebug_17-10-03_17-58-52.log)

 Taranis X9D:  If using one of this Telemetry Fields with Numbers

TX-Voltage
Time
Timer1
Timer2
Timer3

Companion Crashes when calling Telemetry Picture with Page Long.

All Working Telemetry Fields are Displayed with TELEx: "Name of Field"
( Sorry in the Moment i can not test at the real Tx....) 
With N388 was OK, not tested the Versions between.
Ralf

schwabe, 2017-10-03:
Can you include a saved otx file that crashes when you press simulate? That would make debugging for us much easier.

r41065, 2017-10-03:
[W3.otx.zip](https://github.com/opentx/opentx/files/1353229/W3.otx.zip)

maybe you make .otx Files zo Upload directly  

Ralf

r41065, 2017-10-03:
Simulator starts normal, crashes in the Moment i switch to Display the Telemetry Page with PAGE Long...

schwabe, 2017-10-03:
Hm works fine here:

[image]

kilrah, 2017-10-03:
X9D, not X9D+... can reproduce here...

schwabe, 2017-10-03:
hm cannot reproduce on my mac.

kilrah, 2017-10-03:
And I can't properly debug on windows since for some reason the simulators never load when built in DEBUG mode :(

powercycle, 2017-10-04:
can't reproduce on Ubuntu 16.04, but not built with debug...mine is from PPA.

r41065, 2017-10-04:
[image]
[image]
Second Screenshot shows my working Lua Script

Update N396:
Value fromTimer3 is more than 11 hours, 41 Minutes, if this will be deleted from Page the  Simulator didn't crash but only shows strange Values for Batt, Time and Timer 1...

r41065, 2017-10-22:
[image]

Update N412:
Simulator didn Crash anymore, only Display Strange Values on Battery Voltage, Timer 1 and Time...

3djc, 2017-10-22:
Have you not rediscovered sensors and they are now all shifted ?

r41065, 2017-10-22:
because i'm in  China since a few weeks i can only open my otx-files from Version 2.2.0
therefore i said that i cannot test with my real TX.
i do not discover new sensors inside Simulator....

Headline from Telemetry Page and my Lua Screen show the right Values, so the Data itselve seems to be Ok

3djc, 2017-10-23:
There is indeed something very broken, will have a look at it

r41065, 2017-10-28:
Update N418 / N419
Simulator Crashes every First Time after new Installation of Companion when i call the Telemetry Page with Timer Values,
After that it shows the Strange Values..
Sorry but i cannot really find a pattern in behaviour

Ralf

3djc, 2017-10-28:
The fix has not been merged yet, so it is not yet available in nightlies

r41065, 2017-10-29:
Ok

closed 2017-10-29 by kilrah

r41065, 2017-10-31:
[image]

Update N422:
- Simulator did not crash anymore
- Values for Timer and Time are displayed correct

- Value from Battery Voltage shows same strange  behaviour as before, 

may i reopen this point or will there be an solution ?

Ralf

3djc, 2017-10-31:
It is even stranger, it crashes the radio, so yes , there will be follow ups

3djc, 2017-11-07:
I just merged the fix, should be ok tomorow

PR #5299 (2017-10-29): Fix timer display (is UNIT_DATETIME but no date available)
This fixes #5263 (the later part, a bug I introduced)