opentx/opentx #5264: Horus: BOOL type in options declaration not of type boolean

lshems, 2017-10-08:
It seems that the BOOL type used in the options declaration to get a checkbox in the widget options dialog is not an actual boolean, but toggles between 0 and 1 instead.

This is confusing, and could perhaps better be changed to a real boolean type??

schwabe, 2017-10-08:
hm. I am not a lua expert but wouldn't that break all existing scripts?

lshems, 2017-10-08:
sure, all the widgets using this boolean. But this isn't nice either.

mpaperno, 2017-11-20:
> not an actual boolean, but toggles between 0 and 1 instead.

Well that's how booleans behave in pretty much everything except Lua. Maybe it could be called "CBOOL" to avoid confusion.  Complain to whoever decided that `0 == true` was a [good idea](https://www.lua.org/pil/2.2.html)...  😉 

I'm not sure you could use `true`/`false` keywords to set the default option values either, unless `lua_tounsigned()` treats those properly as 1/0 (haven't tried).

```cpp
            else if (option->type == ZoneOption::Bool) {
              luaL_checktype(lsWidgets, -1, LUA_TNUMBER); // value is number
              option->deflt.boolValue = (lua_tounsigned(lsWidgets, -1) != 0);
              // TRACE("default bool = %d", (int)(option->deflt.boolValue));
            }
```

commit bb460089 (2019-11-08): Fixes #7059
Fixes #5264
closed 2019-11-08 by bsongis