opentx/opentx #5268: Telemetry: consumption

Allerhopp, 2017-10-10:
DIY current sensors allow to measure  bidirectional and in mA.

There are also two units for current available: A and mA. Consumption always calculates with A, is ist possible, to look for the unit of the current sensor and to adjust the consumption calculation in case of mA?

Consumption adds the absolute value of current, e.g. a solar plane battery can be charged in flight, is it possible, to account the sign of the current in the consumption calculation?

[image]

kilrah, 2017-10-10:
FrSky telemetry spec does not support negative current values.

Allerhopp, 2017-10-11:
Sure, but the flexibility of OpenTX allows to use bidirectional sensors and the unit mA. Then the calculation of consumption is simply wrong. But i have no clue, how much work it is, to change the routines. It would be great, if the routines were more universal. Please feel free to close this issue, if there is no chance for a change.

kilrah, 2017-10-11:
If you change the unit to mA in the sensor settings it is only a display thing. The sensor must still send data in the same format (A/10). 

Not enough context and info about what kind of sensor you'd be wanting to use to give more insight.

Allerhopp, 2017-10-11:
OpenXsensor supports the ADS1115, a 16bit A/D converter. It has 4 groundbased inputs or 2 differential inputs. It has already been used for analog airspeed, voltage and current measurement with an analog current sensor. It has gain amplifiers, that allow 256mV or +-256mV (diff. input) measuring range. Perfect to use for current measuring with a shunt. With a 1mΩ shunt 256A range for example. It can also measure small currents eg. RX, servo etc. With a 50mΩ shunt 5A range for example, with a resolution of < 0.1mA. In this case it is better to transmit in mA, to keep the resolution. 
It is possible, by misusing a TEMP or ACC field, to transmit, show and  log mA, because OpenTX offers this unit and the name of the sensor can also be easily changed. I did it in my example above.
OpenTX can also calculate consumption for this sensor, but then things start to go wrong. OpenTX presupposes that A is the only used current unit and negative current is never happening.

This application is only possible, because OpenTX is very flexible and i am very happy with what has been achieved right now. A proper consumption calculation would make this usage perfect. Users of the standard current sensor will never notice the changes, except they change the unit from A in mA.

SMD shunts are small and can easily be integrated in every cable.
[image]

Allerhopp, 2017-10-12:
I realised, that the calculation in this special cases can also be done with a LUA-script and the results can be send to the telemetry and can be logged.

closed 2017-10-12 by Allerhopp