opentx/opentx #5270: X10s

meoster, 2017-10-12:
Powering down will lead to getting stuck contineously in emergency mode.  

I would need to plug a usb cable in and fool it back to bootloader mode.  After that it can boot in properly.  But will still have the power down issue.

bsongis, 2017-10-13:
2nd report of this issue
I cannot reproduce here, unless ... what are your options (the ones you checked in Companion)?

meoster, 2017-10-13:
I have tried 403 pre-built nightly with allowed nightly checked.  It is using 403 nightly and SD v11.  
I have also tested from building the hex source.   I even enabled CLI and debug to put traces around the area.  It seems to be just before the final pwr_off call where the normal  shutdownReason = NORMAL_POWER_OFF is set.   could be something around that time it deadlooped...  and never reach pwr_off. I even tried to move pwr_off up earlier.  that didn't seem to work.

meoster, 2017-10-13:
I just ran it again.  It never made the call to SdDone().

I tried commenting those below also. It still didn't call.  So likely something in the background. 

  }
#endif
TRACE("Setting unexpected shutdown to 0");
  g_eeGeneral.unexpectedShutdown = 0;
  storageDirty(EE_GENERAL);
  storageCheck(true);
TRACE("Unexpected shutdown after storage %d",g_eeGeneral.unexpectedShutdown);
TRACE("Storage checked");
#if defined(CPUARM)
  while (IS_PLAYING(ID_PLAY_PROMPT_BASE + AU_BYE)) {
    CoTickDelay(10);
  }
  CoTickDelay(50);
#endif
TRACE("SD Card");
#if defined(SDCARD)
  sdDone();
#endif
}
#endif

****************CLI Log *********************
>testdsd
SD card has 62333952 sectors
Starting single sector read test, reading 16 sectors one by one
sector 62333936 read OK
sector 62333937 read OK
sector 62333938 read OK
sector 62333939 read OK
sector 62333940 read OK
sector 62333941 read OK
sector 62333942 read OK
sector 62333943 read OK
sector 62333944 read OK
sector 62333945 read OK
sector 62333946 read OK
sector 62333947 read OK
sector 62333948 read OK
sector 62333949 read OK
sector 62333950 read OK
sector 62333951 read OK

Starting multiple sector read test, reading two sectors at the time
sector 62333936-62333937 read OK
sector 62333938-62333939 read OK
sector 62333940-62333941 read OK
sector 62333942-62333943 read OK
sector 62333944-62333945 read OK
sector 62333946-62333947 read OK
sector 62333948-62333949 read OK
sector 62333950-62333951 read OK

Starting multiple sector read test, reading 16 sectors at the time
sector 62333936-62333951 read OK

>opentxClose
luaClose 0xd0109258
luaClose 0xd00ff640
Flush current model
Setting unexpected shutdown to 0
eeprom write general
writeFile(/RADIO/radio.bin)
disk_write bad alignment (0x10002640)
disk_write bad alignment (0x10002640)
Unexpected shutdown after storage 0
Storage checked

meoster, 2017-10-13:
ok I tried again commenting those cotickdelay between..

it went further but till  went into emergency mode. 

Here is the other part of the code. power switch release didn't get call this time. 
void boardOff()
{
  BACKLIGHT_DISABLE();
  TRACE("Boardoff");

  while (pwrPressed()) {
    wdt_reset();
  }
  TRACE("Power Switch released");
  SysTick->CTRL = 0; // turn off systick

  pwrOff();

}

***********CLI log *************
>eeprom write general
writeFile(/RADIO/radio.bin)
disk_write bad alignment (0x10002638)
disk_write bad alignment (0x10002638)
opentxClose
luaClose 0xd0109258
luaClose 0xd00ff640
Flush current model
Setting unexpected shutdown to 0
eeprom write general
writeFile(/RADIO/radio.bin)
disk_write bad alignment (0x10002640)
disk_write bad alignment (0x10002640)
Unexpected shutdown after storage 0
Storage checked
SD Card
sdDone
Boardoff

meoster, 2017-10-14:
Tested 405 nightly.  Problem disappeared..

meoster, 2017-10-14:
It came back after I tried to setup the tx.   will try to debug it again.

agnauck, 2017-10-20:
Same issue on my X10S with nightly build

ducatiJake, 2017-10-26:
I am using the nightly build 2.2.1N416 10/25 on an X10, not an S and also see this problem pretty frequently.  I have noticed that when it happens, if I hold the PgU/D and power down or the Select button and power down, it will often stay powered down.  File I used is opentx-x10-ppmus-noheli-bindopt-en-2.2.1N416

putimir, 2017-10-29:
Same here, i have used all versions from 400-409...

The problem appears from time, but frequently. The only way sometimes to shut it down is to plug in the usb cable, it the goes into dfu mode...

buzzzsam, 2017-11-06:
Testing on it again.. Really want to get this to work properly. 

https://youtu.be/lixPW8Y6ZSA

agnauck, 2017-11-06:
similar behavior here. But when it happened on my X10s in the past I was always able to switch off completely with a long power button press while it was in emergency mode. Have not seen the loop in emergency mode yet.

Will spend some time today and see if there is a specific pattern to reproduce. My first idea was that it could be related to the press duration of the power button when trying to switch off.

buzzzsam, 2017-11-06:
Just went out to buy an 8Gb slower class 10 SD card.  I am able to shutdown on the emergency mode. 

So in summary, 
SanDisk 32GB extreme will fail 100% of the time. 
SanDisk 8G Ultra will fail to emergency mode, but will be able to shutdown after....

May be a SD card driver issue?

bsongis, 2017-11-11:
I just tested again with this bin:
http://downloads.open-tx.org/2.2/tests/opentx-x10-multimodule-noheli-luac-en.bin
Impossible to reproduce.
Does the bug happen just after you make a change (I tested this as well, but just in case ...)?
How many ~ tests do you do before reproducing the issue

meoster, 2017-11-11:
It happens very frequently for my unit.   I can take some video.

kilrah, 2017-11-11:
Did someone try changing SD card?

buzzzsam, 2017-11-11:
https://www.youtube.com/watch?v=-RWGDutTZEg

Here is recent source compiled version using an 8G card. 

I will run the test bin again and test.

buzzzsam, 2017-11-11:
https://www.youtube.com/watch?v=ehN7YDr1Td0

This is using the test bin from Andre.

buzzzsam, 2017-11-11:
Just tried to debug github tree.  

Getting lots of 
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)
disk_write bad alignment (0x1000ae48)

Not sure if it is relevant.

buzzzsam, 2017-11-11:
Enabled the detailed write trace

disk_write 0 0x20003e40    1320498 1
disk_write 0 0x1000ae48    1320499 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320499 1
disk_write 0 0x1000ae48    1320500 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320500 1
disk_write 0 0x1000ae48    1320501 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320501 1
disk_write 0 0x1000ae48    1320502 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320502 1
disk_write 0 0x1000ae48    1320503 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320503 1
disk_write 0 0x1000ae48    1320504 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320504 1
disk_write 0 0x1000ae48    1320505 1
disk_write bad alignment (0x1000ae48)
disk_write 0 0x20003e40    1320505 1
disk_write 0 0x1000ae48    1320506 1

kilrah, 2017-11-11:
Yeah those are normal

RoninMedia12, 2017-11-12:
I just saw this thread just opened a new issue for this as well for n433  Mine goes in to cycle and only way to get it to stay off is long press in emergency while plugged into pc or it will just keep cycling

RoninMedia12, 2017-11-12:
Changed to dark bmue theme and issue seemed to go away if i switch back to defualt theme it returns

buzzzsam, 2017-11-12:
Confirmed, changing theme made shutting down worth 9/10 times.  I did get one emergency during my test.  Will try to cut out some animations to see if that's the problem.

3djc, 2017-11-12:
9/10 is still an issue, but something we can look at for clues

bsongis, 2017-11-12:
does the flag "Check Rssi on Shutdown" change anything to this issue?

buzzzsam, 2017-11-13:
Will check

buzzzsam, 2017-11-13:
Just tested.  It still goes into emergency mode, but seems like I am getting more successful shutdown with the rssi check off and removing the sleep bmp.  like 2/10 fails.  Much better than before.

raphaelcoeffic, 2017-11-13:
Strangely, I had the issue right after installing OTX on a brand new X10. After a couple of power up/down, the issue was gone.

kilrah, 2017-11-13:
If I understand well @buzzsam can build firmware, can you try my branch `kilrah/5270_wdt_test`?

buzzzsam, 2017-11-13:
Yes will test

buzzzsam, 2017-11-13:
Tested, still cannot shutdown 100%.

putimir, 2017-11-15:
Same here. I tested Sandisk Ultra 32BG, 16Gb, 8Gb, noname  2Gb. Still sporadic emergency mode. I can't seem to shutdown from "Em" eather...

kilrah, 2017-11-16:
Are all reports here with the X10**s**, or some with X10 as well?

raphaelcoeffic, 2017-11-16:
@kilrah I had the issue a couple of times right after first install on a X10, and since then, nothing. Maybe something with the default settings, that get overwritten once you've played a bit with the TX.

putimir, 2017-11-16:
Mine is X10S

buzzzsam, 2017-11-16:
May be someone can teach me how to link up the debugging on it. I can do a step trace after powering down.

kilrah, 2017-11-16:
https://github.com/opentx/opentx/wiki/Taranis-Hardware-Debugging

ducatiJake, 2017-11-16:
I have been seeing this since I got my X10.  Not an S.  I have updated the firmware several times with no change.  I can usually get it to shutdown from the emergency screen if I hold down the selector PgUp/Dn button while powering down. Sometimes it takes several attempts.  The longer I leave it on and more I do the more likely it is to go into emergency mode.

kilrah, 2017-11-18:
Can a few people with the problem try this bin? http://www.andrebernet.ch/divers/opentx/opentx-x10-shutdowntest1.bin

buzzzsam, 2017-11-19:
Just tested 10 times  10/10 shuts down properly.  Amazing job !!

putimir, 2017-11-19:
I hope it's not too early, but same here! 👍

Also noticed, that to power the TX on, you have press and hold the power button for, lets say 100-200ms, before this bin it was instantaneous.

kilrah, 2017-11-19:
Nice that the reason is confirmed, not nice that it means we'll have to remove the safety mechanism that allowed to catch ESD-related resets and reboot to emergency instead of shutting down for good. 
This appears to have been well prevented in hardware now so hopefully not too bad.

kilrah, 2017-11-20:
Can someone with the problem test the hopefully final fix? http://www.andrebernet.ch/divers/opentx/opentx-x10-shutdowntest2.bin

putimir, 2017-11-20:
10/10 successful shutdown with my X10S...woohoo :) 

👍

ducatiJake, 2017-11-20:
Been running it for a few hours now with no hangups on my X10.  I noticed the delay at startup as reported before.  Not an issue, just different.  Nice work, good progress!

buzzzsam, 2017-11-21:
working!

closed 2017-11-21 by bsongis

BladeScraper-Designs, 2019-05-20:
Same issue here. Though not only does it enter emergency after turning off (and it reboots), I can hear the trim sound when holding the power button to power it down.
https://youtu.be/H27wtpO7-co

Tried reflashing firmware, changing SD card, stock SD card content, etc.  Nothing helps.  Only way to boot normally is to  unplug the battery and plug it back in.

commit 39ea315c (2017-11-21): Fix X10 power loop (fix #5270) (#5391)

* Fix X10 power loop (fix #5270)

* "EMERGENCY MODE" text wasn't visible anymore since d4a00dec3b3bb643bb322a1088812f4f9b908404