opentx/opentx #5271: X10s default style background color change slow

meoster, 2017-10-12:
Color change for default style background color changing is very slow

meoster, 2017-10-14:
I found out why it is slow.. for the background color change in user interface.  A rambackupwrite is done for ever dial change.  this likely added lots of write lag into the menu.  For a similar control in the channel widget.  The rambackupwrite is down after the dial has been selected and returned.  

>trace on
>RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
RamBackupWrite sdsize=10227 backupsize=4531 rlcsize=452
eeprom write general

meoster, 2017-10-14:
Disabled the rambackupwrite to test, got a little faster; but something else is still causing the lag.

rcbebo82, 2017-10-25:
Same as #3799

kilrah, 2017-10-29:
Closing as duplicate.
The reason is slow is having to render the entire screen again (instead of just using the cache).

closed 2017-10-29 by kilrah