opentx/opentx #5275: LUA environment issue's: can't access the constants in the _ENV, though they are global?

lshems, 2017-10-14:
in openTX LUA we don't have the _G global variable accessible.  We do have the _ENV environment variable available.
However:

the available constants SMLSIZE, etc. are not part of the _ENV variable. This is inconvienient, if you want to know them, by looping through them:

for k,v in pairs(_ENV) do
  print(k,v)
end 

and check the result in the debugger.

for instance the lcd.drawText(0,0,"testtext",256) will produce a very tiny font, but no way to find ut the constant linked to 256.

kilrah, 2017-10-28:
Since that last argument is flags you're never going to get constants out of it. Read the doc...

lshems, 2017-10-29:
I think you misunderstood. It is just to prevent reverse engineering that I want to have a list of available constants. It would even be better if they would be grouped together per type in an array. 

Actually I don't want to know how you manage the flags, but still want to know options to put in.

The example of 256 is what it is: an example. I could as well have used a constant mentioned in the docs. Can't get a list of those either in Lua, causing all sorts of risks, since if I would try to declare a global with the same name, it wouldn't work.

The point is, more generally speaking, that the global variables need to be known to a script or its developer. Documentation is not the preferred way.

lshems, 2017-10-29:
The global variable _ENV  is exactly there for that purpose.

lshems, 2017-11-02:
The 'lcd' and 'model' rotables should be part of the environment as well. Actually all 'named' things should be accessible and visible when checking for them in the global environment _ENV using:
`for k,v in pairs(_ENV) do
 print(k,v)
 end`

projectkk2glider, 2017-11-04:
We don't use `_EVN` or `_G` varables because of the `rotables patch` which saved a lot of memory (ea122610efe0c015251169ff91253ba9088bc059).  Not sure we can do anything sensible about this and not increase RAM usage. I guess you will have to work around this issue.

lshems, 2017-11-04:
Pity, but a good reason. Thanks for the info!

3djc, 2017-11-04:
Closing this then

closed 2017-11-04 by 3djc