opentx/opentx #5276: New curve functions round(x) and int(x)

andreab3, 2017-10-15:
Mapping the input value from a potentiometer into N number of specific values, for example when using a pot as a 6 position switch, currently requires creating a custom curve with N flat areas corresponding to the desired output values and tedious amount of basic math to distribute the areas evenly.

This would be much simpler with the addition of new options under **Curve -> Func** when setting the Curve in an Input:

1. **int(x)**, or just "int":  keep the integer portion of x and discard any decimal portion.
2. **rnd(x)** or just "rnd":  round the value of x to zero decimal places.

With these new functions, we can now map analogue values 0.0 - 100.0 to digital 1 - 6 easily by setting up the Input on pot S1 as:
- Set weight to 5
- Set offset to 1
- Set curve to func -> rnd(x)

Then we can spread the values out across the -100% to +100% range by multiplying the values using weights in nested mixers ( eg; weights of 500 and 200 will multiply values by 10x) so that our 6 position values are distributed across a wider range and can be used as inputs in the BetaFlight GUI.

projectkk2glider, 2017-10-15:
Have you considered using Lua mixer script to get your desired behavior? This would be much more customizable and can work out of the box right now without any (extensive) changes to OpenTX.

schwabe, 2017-10-15:
or even just a lua script that creates those special curves for you

andreab3, 2017-10-15:
I have yet to delve into Lua, and I'm sure it'll open a whole new realm of possibilities once I get into it.  Howerver, this looks like a relatively simple change that makes it easy and quick to do very cool things with OpenTX, without people having to spend a lot of time making custom curves or learning Lua.
As more functionality is introduced to BetaFlight and other FC software, making the most of the available AUX channels becomes a real issue, so I see this feature as part of a bigger solution for making "channel conservation" much easier to implement for everyone.

kilrah, 2017-10-28:
Seems the multipos handling code already does what you want. If not, give a clear example.

andreab3, 2017-10-30:
Perhaps, kilrah, but this is for a regular potentiometer set up as a "Pot", not a rotating switch set up as "MultiPos".  Maybe I didn't state it clearly enough, but the goal here is to be able to _easily_ make a regular potentiometer emulate a Multipos switch, but with any number of stops.

So far I've been doing it with custom curves and have made my S1 act as a switch with 3, 6, or 16 positions, but it takes a lot of time to create those curves and it's quite messy.  I currently have S1 set up as a 21 position switch and can select any of the BetaFlight in-flight Adjustments (plus Beeper!) using S1 alone and a single channel.

This is all only possible with the awesome power of OpenTX (!) but it would be so much easier with these int(x), rnd(x) functions.

kilrah, 2017-10-30:
So if I understand well you want to reduce resolution and create steps?
Easy to do using the mixer, and with any step size you want.

[image]

closed 2017-10-30 by kilrah

powercycle, 2017-10-30:
@kilrah, doing what you are suggesting above would be similar to a curve with the needed number of points?

Just accomplishing the same with the mixer, is that right?

3djc, 2017-10-30:
Yes, just pick the values that best match how many steps you want

andreab3, 2017-10-31:
Maybe I'm missing something obvious, but the example you gave, kilrah, doesn't work for me.  I'm using OpenTx 2.2.0 (63baa76c), dated 2017-09-05 20:07:19 that came with my Taranis Q X7S.

When I try the example (which essentially divides the signal by 10, then multiplies it by 10 again) and watch the output, I just get the same signal that goes from that goes the full range of 989us to 2011us on my Taranis Q X7S.  There are no steps, it progresses 1us at a time.  It's basically a x=y line.

Is setting the weight to 10% supposed to truncate the decimal part of the value?  It doesn't do that for me here.

kilrah, 2017-10-31:
Yes, worked for me on the simulator, with steps of 1% instead of the normal ~0.2%.

3djc, 2017-10-31:
As a side note, a LUA to create a X point curve is trivial to make, and would be very easy to share within the community, and also, if you want 16 values, it starts to be tricky with a pot, I would use a unused trim switch instead personnaly (but I was sidetracking)