opentx/opentx #5281: V2.2.0 Trims are Reversed on Primary Control Function when Reversed on Input Screen

duartecabral, 2017-10-15:
Nearly CRASHED my 50cc Airplane because of this issue!!!! 
If Ailerons or other controls are reversed on the Input screen, the trims are not reversed along with the primary control so that they match. This means that when flying, applying trim in one direction for making a trim correction means that the airplanes response is to worsen the trim condition! One better be good at doing rolling circles till one figures out that the trim is backwards! and act accordingly!!!  And if one uses extended trims as I did, the trim induced roll can be very fast!

hrenz, 2017-10-17:
Yes, a long known "issue" or behavior since years
You can reverse Inputs, for example ELEvator, but not reverse TrimELEvator

3djc, 2017-10-17:
You can choose the behaviour by choosing  you apply your -100 to either input or mixer

closed 2017-10-17 by 3djc