opentx/opentx #5286: V2.2.0 Throttle Trim Graphic and Trim Do NOT work as expected with Extended Trims

duartecabral, 2017-10-15:
If extended trim is selected for the throttle, the servo only moves from the mid point to high with the throttle stick movement. To get the lower part of the throttle servo movement one has to move the throttle trim to the bottom of the range. (This is not what one expects to happen. I expected that I still got full servo movement with throttle stick movement when Extend trim was selected. This has the effect the graphic showing the trim position is bottomed out and one has to rely on the number that is reported which defeats the purpose of the graphic.
Would like to have, when extended trim is selected, trim graphic in center, and servo movement -90 to +100 with stick movement. Extended is would then just a greater movement of servo for trim movement. Bottom of trim becomes 125% instead of 100% (and so mid and top of trim gets 90% and 80% respectively or something similar.)

kilrah, 2017-10-28:
> Extended is would then just a greater movement of servo for trim movement.

That's what it is. You may be confused with how processing flow works in OpenTX.

kilrah, 2017-11-08:
Closing for lack of details.

closed 2017-11-08 by kilrah