opentx/opentx #5287: Taranis X7 "FW update error bottom pin no resp"

rrlarenas, 2017-10-16:
Hello guys, well I got a newer taranis X7, but i'm uneable tio flash my rx using the bottom s-port in the tx.. I've got this message "FW  bottom pin no resp" the RX turn on just  a fraction of sec then,it turn off and i've got the message on the radio..

I'm Ussing O-TX 2.2.0

Any help?

schwabe, 2017-10-16:
Double check everything. Note this is a bug tracker. Not a help forum. Use rcgroups or openrc forums to seek help.

closed 2017-10-16 by schwabe

rrlarenas, 2019-01-15:
> Hello guys, well I got a newer taranis X7, but i'm uneable to flash my rx using the bottom s-port in the tx.. I've got this message "FW bottom pin no resp" the RX turn on just a fraction of sec then,it turn off and i've got the message on the radio..
> Hi rrlarenas,
> Were you able to find a solution to this problem? I have the exact same problem.
> Look forward to your help and reply.
> Thanks

Hello there 

Just update O-Tx to last versión and it works

Get luck!!

Atte.

Rodrigo Larenas

3djc, 2019-01-16:
It does work, You are either using the wrong cable (you need a simple straight one) or have it reversed

kilrah, 2019-01-16:
Ot the receiver is faulty, we heard a few times of receivers that refused to be flashed.