opentx/opentx #5289: Throttle warning not disabled when using CH1 as throttle source

JapieG, 2017-10-17:
I am not sure if i am using it wrong or if i encountered a bug.

I have throttle on a switch. In the model menu i can not select a switch as the throttle source (enhancement? lot of glider pilots using switches for throttle). So i select CH1 as my throttle source. This is off course the channel where my ESC is. 

I am expecting that when the value of CH1 is -100 the system indicates the throttle as OFF, so also the throttle warning on bootup should have been cleared. It does not. In fact, i need to move my throttle STICK to an end position to clear the warning. I don;t even have my throttle stick bound to anything throttle like in my model file.

OpenTX n400 on X10s

If i m using it wrong and this is not a bug please close the issue with a smalle explanation on how to use it correctly. Tnx

3djc, 2017-10-17:
Throttle source is not as flexible as we would like it yet, and cannot cope with your situation yet. What you need to do is disable throttle warning, and set a switch warning in the right position.

it is something we want to improve at one point in time

closed 2017-10-17 by 3djc

JapieG, 2017-10-17:
Yes i did that! Thanks for the explanation. Just thought i saw a bug because of the warning still reacting to stick input unless the stick is nowhere selected as throttle. (except maybe in the STICKMODE selection. Thanks!)

3djc, 2017-10-17:
You can disable throttle warning here : 
[image]