opentx/opentx #5290: Q x7s bricked

ivv72, 2017-10-17:
Hello,

I wanted to upgrade my taranis x7s with open tx firmware and I only found firmware for the Q X7, andnot for the Q X7S.
So I was stupid enough to take the Q X7 firmware and flash my Q X7S with it.
The consequence was that I bricked my taranis.
It turns on and shows the splasch screen and when the throttle warnig goes the screen goes blank.
Can any body help me, onthe frsky site there is only firmware forthe Q X7 and also not for the Q X7S.
Thank you

greetings

Igor

andreab3, 2017-10-18:
Odd, when I tried that it recognized that it was a Q X7 and didn't let me flash it.  Anyways, if you email me at temp123@audasys.ca I can send you the backup I made _before_ I tried to flash it.. :)

geofrancis, 2017-10-23:
you should be able to start in bootloader mode and reflash the firmware

3djc, 2017-10-23:
Not an OpenTx bug, closing it

closed 2017-10-23 by 3djc