opentx/opentx #5291: Bricking taranis Q X7S

ivv72, 2017-10-17:
Hello,

I wanted to upgrade my taranis x7s with open tx firmware and I only found firmware for the Q X7, andnot for the Q X7S.
So I was stupid enough to take the Q X7 firmware and flash my Q X7S with it.
The consequence was that I bricked my taranis.
It turns on and shows the splasch screen and when the throttle warnig goes the screen goes blank.
Can any body help me, onthe frsky site there is only firmware forthe Q X7 and also not for the Q X7S.
Thank you

greetings

Igor

schwabe, 2017-10-17:
Firmware for X7 works on X7S. Note we only support the version that are downloaded through OpenTX companion and not those on the opentx homepage

ivv72, 2017-10-17:
Hello

I downloaded the firmware through open txcompanion so now I dont know what the problem is.
I think it started when I tried to import models from a X9D.
Any ideas anyone?

schwabe, 2017-10-17:
directly importing the x9d models on the tx will not work, you need to this with Companion

ivv72, 2017-10-17:
Yes it needs to convert the files, and on the X7S there are fewer switches so you get some errors.
At first it would not take my models but later it seemd to do so.
And then the trouble started

schwabe, 2017-10-17:
@ivv72 If you imported directly x9d+ files on your x7 you might need to reset eeprom to get a working state on your x7 again.

closed 2017-10-17 by schwabe

ivv72, 2017-10-17:
Would you be so kind and tell me how I can reset the eprom?
I would be very thank full
Thank you