opentx/opentx #5292: V2.2.0 Trims are Reversed on Primary Control Function when Reversed on Input Screen

duartecabral, 2017-10-17:
Response by 2 individuals
hrenz (response)
Yes, a long known "issue" or behavior since years
You can reverse Inputs, for example ELEvator, but not reverse TrimELEvator
(The above is the experience of the author of the issue stated earlier, the trims are backwards to what was reversed (-100 on the input screen!)

3djc (response)
You can choose the behaviour by choosing you apply your -100 to either input or mixer

Response to Issue Closing:
Apparently "3djc" was unable to comprehend the issue as stated (but "hrenz" did as is quoted above.) and so "3djc" closed the issue. This issue is significant as it will cause many airplanes to crash!!!! and so this issue needs to be corrected!!! Again: Applying left Aileron Trim causes the airplane to role right!!! Which is wrong response, but moving the Aileron stick to the left causes airplane to role left which is the correct response.  Please correct this issue as will cause many crashes.

The initial issue is as stated below: 
Nearly CRASHED my 50cc Airplane because of this issue!!!!
If Ailerons or other controls are reversed on the Input screen, 
the trims are not reversed along with the primary control so 
that they match.....

jtaylor2, 2017-10-17:
I think 3djc understands the issue very well.  The problem is you are applying the negative rate in the wrong place for what you are trying to do.  This is working as designed.  Negative weights in inputs only reverse the sticks, not the trims.   You should not use negative weights in inputs to reverse servos.  You should use either negative rates in mixes (where it will also reverse the trims) or use positive rates in inputs and mixes and reverse the servo in outputs.  This is the way it has always been since the inputs screen was first created and is working as designed.  Only use negative rates in inputs if you want the trims reversed or select no trim (in inputs) and add the trim back in mixes.  If all you want to do is reverse a servo it should probably best be done using the reversing option in outputs.

schwabe, 2017-10-17:
Also please don't open new issues on existing issues. You can still comment on existing issues. I am closing this as like @jtaylor2 explained, everything is working as designed.

closed 2017-10-17 by schwabe

duartecabral, 2017-10-18:
It would be most helpful if this explanation of how the trims work with respect to the primary controls in the "input window"  and "Mix window" would be contained in the documentation. Having been in the hobby for over 40 years, the general expectation is, if a channel is reversed, the trims are also reversed as they are a set that control the same control surface. It is my hope that those that discover this relationship (as claimed "everything is working as designed" between the primary channel and trim while the airplane is in flight will have a quick mind to figure out that the trims are reversed and then apply opposite trim to correct for the issue instead of applying more trim which will exacerbate the issue and result in .... 
I do not think from the 2 rejections of this issue and with the statement "Everything is working as designed" that it will be changed in the future, but please put it in the documentation or a message on the Input screen that if controls are reversed that the trims are not reversed so to give the user an idea of what to expect. 

Other than this disappointment, I am impressed with the software and work that has been done.

duartecabral, 2017-10-18:
Back Ground of how this was discovered. 
I copied an existing model into  new one. I then found that due to servo installation that the 4 servos on the wing (2 on each half)  where all moving the surfaces backwards to what was needed. So instead of going into each servo set up (Mix or Output Screen) I went into the inputs and reversed the ailerons there in one simple change. Unknown to me was that the trims now were backwards until the maiden flight. Almost had a ouch!!!

Just a little clarification as to why am passionate about this issue.

Mow06, 2017-10-18:
IMHO anyone who flies a new 50cc model without checking ALL the controls work as expected before takeoff should go and find a new hobby before they injure someone.

jtaylor2, 2017-10-18:
I actually thought it was documented, but I just looked at the 2.2 manual and don't see anything about it.  You say "the general expectation is, if a channel is reversed, the trims are also reversed as they are a set that control the same control surface." and that is correct.  However, setting a negative weight in inputs is not reversing a channel, it is reversing an input.  And that input can be a stick, a switch, or even a trim.  If you look at a diagram of the input flow https://goo.gl/lBPpfM you will see that trims are applied after the weights and offset, whereas in the mixer flow https://goo.gl/o7VqyZ they are applied before.

I think that all the documentation I have seen says to do rates and expo in inputs and reverse servos and channels in outputs or perhaps the mixer, but I agree that an explanation and/or warning about what happens with trims if a negative weight is applied to a stick in inputs is justified.

Why don't you open an incident against the openTX manual https://github.com/opentx/opentx-taranis-manual/issues with that suggestion and post a link to this incident so they know the context.

duartecabral, 2017-10-18:
This is an interesting discussion and I am pursuing it for what I believe is the best interest of the user who may not have the in depth knowledge of the designers of the software. 

One further point I would like to make is that the "Input Screen" and Mixer Screen are not consistent with respect to how they treat Trims!!! if goes to the Mixer Screen with respect to trims all that one sees is a check box. This has the effect to turning trims on and off. Let say for example they are turned on. Now lets say that I place a "-" in the weight area. what happens is that the servo movement is reversed and the stick and trims track each other. Doing this in the Input screen cause the stick and trim to move the servo in opposite directions. So it is clearly seen that the expectation set by the Mixer screen (which is what is expected by the user) is opposite to what happens in the Input Screen. (Yes I do understand the explanation of how things where built and the resulting functionality)

Now Further: 
I have pulled up the input screen on the transmitter and yes there is a weight function, a positive or negative entry can be made. There is also an entry for "Trim" Two of the selections are "on" and "off"  To me this implies that the trim is being associated (carried through) to the input selection (aileron stick) (or what would "on" and "off" refer to?) and with this association then the polarity that the weight selection is giving it in the upper portion of the screen would also apply( it does apply in one case positive- positive . (Makes sense does it not? why would association not happen to both parts, and what would make one believe that only one association would be valid? Given that this is being presented to the user how can it be expected that the user would expect that the trim is not reversed when the weight of the stick is reversed (the only selections are "On" and "Off". (I do understand the explanation as to how things are actually constructed and I did come up with a working solution before filing this issue.)

Also if the inputs are treated as singular individuals (as mentioned in the above response comments) why are they included in the same screen with another input which is primary? Should they not have their own screen where they could then be "reversed" or "track" the input that they are associated with?

One possible solution is to when trims are selected "on" that it could be "on" with "+" or "-" symbol to alert the user as to which way they take affect and the user could then select which ever polarity is wanted. 

One of the things that I have not tested is the other trim selections such as Rudder etc. I am assuming that these selections where intended for cross trimming. As these trims are not directly associated with the primary selection (Aileron stick to Aileron trim) I would of checked for there expected operation as suggested by "Mow06"

I think that we all now have an understanding of each other positions. and I thank you for the explanation of how things were built and work, and so at this time I see no need to further belabor this issue and I will understand if no response is given to this comment. 

I thank you for the time and effort given to the explanation as to the reality of how things were constructed and work, and I hope that others do not have a negative experience on discovering to late that trims are backward to what they expect (from historical experiences that caused an expectation) of how things should work, or from what they have been led to expect how the trims work from the Mixer Screen experiences.

3djc, 2017-10-18:
I understand your message more than you think. If you looked several years back, you could almost find a clone copy of your made by .... me. And I'm flying rather expensive F3A planes, so I understand the fealing.

It doesn't change the bottom point : you did your inversion at the worst possible place. Not only for the reason you mention, but also because all you mixes will have to suffer from it. Let's take a basic exemple , a simple cub like plane, with wing on top. Because of the design of the plane, often, they tend to go up  when you throttle more, so you'll want down elevator when throttle go up. But because you inverted elev on input, you now have to apply a positive value to get down. very confusing.

Your servo inversion need to be done as much as possible in the servo pane, so that all the chain is 'in the right order', only servos mouvement are inverted or not to match physical layout.

Something that did help me at the begining is this : "trims are added after input". Once I understood that, it helped me