opentx/opentx #5295: V2.2.0 "Write Error" on 9XD+ through Companion 2.2

JosuaBlommie, 2017-10-19:
Whenever I attempting to flash Taranis 9XD+ (yes, correct firmware version) through Companion 2.2, I get a "Write Error." However, I can copy the firmware file to SD card and successfully flash in bootloader options by selecting it with the Taranis Buttons. I'm not sure if flashing the firmware through the bootloader performs the exact same function as through Companion, however I would like to change settings and splash screen in companion before flashing. 

Please see the photo of the error message attached. I have exhausted every option and possibility to my knowledge before coming here. Any ideas?? 

[https://drive.google.com/file/d/0B1bV67VUpTjuOXY1MWY4ck1wZG8/view?usp=sharing](url)

JosuaBlommie, 2017-10-19:
Please try the link again, I revised it.

powercycle, 2017-10-20:
Here are my thoughts:  Use a different cable (try a couple of different ones), Use a different USB port, use a different computer.

My assumption is that your machine is Windows.  It has been a while since I've done any hardware troubleshooting on a windows machine, but this definitely seems like a hardware problem.   I haven't yet looked through current issues to see if "Write Error" shows up as a current problem.

I've been using 2.2 for a while now without issue.

JosuaBlommie, 2017-10-21:
Thank you for your suggestions. It seems the writing the firmware through
bootloader works just as well. I'm happy with where my radio is at the
moment, and I don't fancy bricking it again.

Kind regards,

Josua.

kilrah, 2017-10-28:
Bootloader version?

closed 2019-11-15 by stale[bot]