opentx/opentx #5298: Is there a change in the behavior of the reset action in special functions in 2.2.0?

pau33, 2017-10-22:
I am now trying to switch from 2.1.9 to 2.2.0 (X9D+) and are stepping into several problems.
One of that:
- I have a special function, which resets my GPS-Sensor signal by a switch. This reset ist caught by my lua to set an new origin. In 2.1.9  as long es pulled the switch, the GPS-Signal was 0. Now in 2.2.0 it only resets the GPS for a very short time, it will come back, even I have not released the switch. That time is not sufficient for my lua script, yet.

Is this a the new valid behavior or a bug? If it is the new behavior, how can I reset the GPS as long, as I pull my switch?
Thank You in advance.

pau33, 2017-10-23:
From further inspection, it look's to me, as if the reset is so short, that the GPS-Sensor ...
 - is only partially resetted (lat or lon)
 - or resetted and partially populated, when the lua mix script reads it's value in a new round

I resolved this situation for me in lua in a way, that I do not trigger on the reset any longer.
But I think, we should now, whether this behavior is intended in this way, as you won't be able now to reset a value for a longer or desired time-period.

kilrah, 2017-10-24:
The Reset SF just requests a reset at next mixer execution, it's not "continuous". If some telemetry data arrives between 2 mixer runs it's expected that it will be briefly populated before being cleared again.

3djc, 2017-10-24:
Also that process last while the reset condition is active (but as Kilrah mentionned, even duiring that phase there can be valid data for short period of time before it is reset)

pau33, 2017-10-24:
Ok, the question was: Is there a difference between 2.1.9 and 2.2.0? As I said, the behavior is different.
I just compared the simulation between Companion 2.1/2.1.9 and the new 2.2 one.

If I load a simulation file with GPS-values in 2.1.9, set a special function as mentioned with a switch and reset of the GPS, start the simulation, look at the GPS-value, pull the switch, the value will be --- for the whole time pulling it. If I do the same simulation with 2.2.0, when pulling the switch, the GPS-value will switch only very shortly to 0 (or even only lat or lon does) an than comes back again while I was pulling the switch.

So, as I said, I can now handle this new behavior in my lua, but if this different behavior is normal form 2.2.x on, we should mention it somewhere, if not, than it is probably a bug.

schwabe, 2017-10-24:
The behaviour was and is undefined, so relying on a certain behaviour is the bug here. I am not sure what you  are too achieve. Maybe there needs to a way to access more of the internal fields of telemetry.

pau33, 2017-10-25:
@schwabe, maybe it was wrong, relying on that behavior, but it was logical at least for me. Or did you mean, relying on the fact that the GPS lat an lon ist populated at the same time might have been luck?

If I have a special function and a SWITCH. I assumed the function lasts as long as the SWITCH-condition is full filled. That worked for the reset-Action in that way in 2.1 (you can see it in the simulator easily). I used it for example for the auto-offset of a calculated sensor (also others than GPS, for example a relative Altitude) in that way that the trigger/auto-offset was at the end of the SWITCH-condition.

In 2.2 at least with the reset-action the special function triggers only once at the beginning of the SWITCH-Action. If it is so, and if I want to trigger on the end of a SWITCH-condition, is the the suggested  way now to setup a separate logical function, that is activated at the end, and reset/autooffset with this separate logical switch?
I only want to understand that point. If that is the (for me new) behavior in 2.2 and this is clear also for other users we can close the point.

3djc, 2017-10-25:
This is plain wrong. Reset is executed AS LONG as the trigering condition is true in 2.2.

3djc, 2017-10-25:
And I tested on companion too, works as I stated above in simu too

pau33, 2017-10-25:
Am I doing something wrong?

[image]

Video:
[reset1.zip](https://github.com/opentx/opentx/files/1414689/reset1.zip)

The same with calculated sensor value Alt2 in companion 2.2 (radio only partly tested but same behavior)

3djc, 2017-10-25:
Are you testing 2.2.0 ?

pau33, 2017-10-25:
Yes exactly it is 2.2.0 (2017-05-30). Was there an issue since its release, I am probably not aware of?

3djc, 2017-10-25:
Then you should test with nightly, we have already made changes in this area, that will be available in 2.2.1, for a different but somewhat linked issue

pau33, 2017-10-25:
Ok, I will do that hopefully later in the evening.

pau33, 2017-10-25:
Test with the newest nightly 2.2.1N416:

video:
[reset2.zip](https://github.com/opentx/opentx/files/1415883/reset2.zip)

The behavior ist different to 2.2.0:
- GPS-sensor ist mostly 0 during fulfilled SWITCH condition, but not totally (see video)
- Calculated sensor value ALT2 is as far a I can see fully 0 when pulling the switch as it should be.

3djc, 2017-10-25:
As we explained above, you should not expect it to remain at 0. It will be periodically reset, but it will  also get populated by incoming telemetry data, before it gets reset.

pau33, 2017-10-25:
Than it's now the behavior as designed.
Ok, good to know.  Thank you for your quick response.
If there are no further questions, we can close that point.

closed 2017-10-25 by 3djc