opentx/opentx #5301: Lua feature request to improve io.read for reading a line

jfrickmann, 2017-10-24:
I find that it is a bit tedious to write a program for processing text files, because I normally want to read a full line of data rather than a fixed number of bytes. When processing a large log file, I also find that CPU time and memory can be an issue. 
In the standard Lua IO library, io.read can read entire lines from a file. It would be great if we could also have that feature in OpenTX. One way that it could be done, would be to make the <length> argument optional, and if it is not present, have io.read(<file object>) return one line.
Thanks!!
Jesper

RealTadango, 2017-11-13:
Something like that would be great :)

flyingfox79, 2018-02-22:
Hello,
I agree, would be great to read a full line in a text file through LUA api, it would be usefull.
I also see the same issue with the CPU and memory limitation, I’m developping a script to read a text file, I got an error message on a small file, few Kb of a text file ends with a CPU time limit error

RealTadango, 2018-02-22:
@flyingfox79 That will always be a limitation, even with a better api. Reading files you probably need to split it over multiple calls. If you need an example, check my Lua log viewer.

RealTadango, 2019-08-17:
Reply to keep it active ;)

closed 2020-05-13 by stale[bot]