opentx/opentx #5302: PPAs doesn't support Ubuntu 17.04 or 17.10

jw323, 2017-10-25:
The PPAs in question;

https://launchpad.net/~opentx-test/+archive/ubuntu/rel
https://launchpad.net/~opentx-test/+archive/ubuntu/ppa

They only support 14.04, 16.04 (the two LTS, which makes sense) and 16.10 (which is no longer supported). Can the PPAs be updated with the latest 17.04 and 17.10? Or at least 17.10 since thats the latest?

I contacted the maintainers of the PPA via launchpad and they said to open an issue on github to add the newer releases.

Thanks

projectkk2glider, 2017-11-10:
@jw323 new releases are available in nightly PPA https://launchpad.net/~opentx-test/+archive/ubuntu/ppa/+packages Can you test them?

closed 2017-11-10 by projectkk2glider

jw323, 2017-11-10:
I was able to install the nightly with the PPA. Thanks!

Could the stable release be updated as well?

Thank you

commit 69afa9d0 (2017-11-10): Fixes #5302: added `zesty` (Ubuntu 17.04) and `artful` (Ubuntu 17.10) and removed `yakkety` (Ubuntu 16.10)