opentx/opentx #5304: X10 trim goes opposite way it should

ducatiJake, 2017-10-26:
I am using build opentx-x10-ppmus-noheli-bindopt-en-2.2.1N416 and am impressed how much works already on this new radio.  I have run into an odd problem where when trimming an airplane the trims go in the opposite direction as expected, making trimming awkward till I figured this out.  I am using a model which I have had for some time, initially creating it on an X9D, ported to an X7, then finally to an X10 I just got.  Trims go in the correct direction on both previous radios, but in the opposite direction on the X10.  The aileron input has a -100% weight, then I have a mix with another -100% weight, then to the output.  Not sure why I did it that way, but that is the way it is set and I see opposite trims.  On my elevator, it is much simpler, 100% on the input, 100% on the mixer then to the output.  Elevator trim works as expected.

3djc, 2017-10-26:
If you have -100 in input, trains WILL be reversed, on all opentx radio

ducatiJake, 2017-10-26:
Checked on my X9d and you are correct.  I guess on the other radios, I just never needed much trim and never noticed.  Thanks

closed 2017-10-26 by ducatiJake