opentx/opentx #5311: Min/Max settings of GVs seem to have no effect

pindeslandes, 2017-10-30:
Companion 2.2.0 simulator, Taranis X9D

1. In the pop up menu of GV1 settings, I set MAX to 50.
2. I set a special function: adjust GV1 with S1 as source.
3. When I move S1 from -100 to 100, GV1 moves accordingly from -100 to 100. Shouldn’t it be clipped to 50 or even scaled to 50 in the positive part?

kilrah, 2017-10-30:
Needs checking but I believe that only affects the increment / decrement function.

elecpower, 2017-11-09:
GV Precision also not applied

bsongis, 2017-11-13:
Testing the whole function now ...
- works correctly with Incr/Decr

bsongis, 2017-11-13:
- with a constant the precision could perhaps be displayed (it is used) and max taken into account (it's not)

bsongis, 2017-11-13:
- with a mixer source, min / max are not taken into account

commit 9eb88e7c (2017-11-13): Fixes #5311
commit e5b2b8ef (2017-11-13): Bsongis/issue 5311 (#5354)

Fixes #5311
PR #5361 (2017-11-18): Companion gvar refactor for special functions to match radio
Fixes #5311
closed 2017-11-13 by bsongis