opentx/opentx #5312: Horus x10 + night building  N421 : Statistics go to rescue mode

nicolas81, 2017-10-30:
When i select "Statistics"  ( after a long press on enter)   my X10 go to rescue mode.

kilrah, 2017-10-30:
Any more details? Can't reproduce.

nicolas81, 2017-10-31:
I'm testing opentx-X10-2.2.1N421 fr
I don't know what kind of details can help you more. I can reproduce this bug everytime. You can see my video :
https://youtu.be/8M65tKr0_78
Thanks for your really good work.

kilrah, 2017-10-31:
Well for a start that you're using the French version, now we can reproduce...

Barbaryann, 2017-10-31:
Hello, The same for me, french version and Statistics mode go to rescue mode...

Barbaryann, 2017-10-31:
Works in the English firmwar version... 
Bug to fix in the french version :D

nicolas81, 2017-11-06:
The bug still remain in version N426 fr

3djc, 2017-11-06:
Which is why it is still opened !

bsongis, 2017-11-11:
I can't reproduce with EN version. Testing now with FR ...

bsongis, 2017-11-11:
Reproduced with FR

bsongis, 2017-11-11:
I think I have it

closed 2017-11-11 by bsongis

nicolas81, 2017-11-12:
Well done and thanks! You're the best.
(I try to understand this code. I installed Qt creator and I managed to compile companion but I don't find a lot of information to understand this code. Have you got some advises to give me?)

commit f6309138 (2017-11-11): Fixes #5312 (#5345)