opentx/opentx #5314: X10: LCB brightness bug

x10user, 2017-10-31:
Bug: Screen backlight
Category: Firmware
Transmitter: tested on X10 (EU) (affected is most likely also Horus 12s)
Tested with: opentx-x10-multimodule-eu-lua-luac-bindopt-en-2.2.1N421
found by: Danny Koch, homo.faber@gmx.de

Description: It is not possible to set the brightness off value so the screen dims completely. It seems the minimum value is about half dimmed. With the FrOS Firmware I was able to dim it completely.
A possible root cause might be the minimum value expexted from the hardware is -100 (instead of 0). This corresponds to the pot input values (possible to assign in FrOS).
The bug is not critical but increases power consumption and reduces operation time on a battery charge.

[image]

kilrah, 2017-10-31:
Intentional, backlight lower than 46% causes display corruption.

closed 2017-10-31 by kilrah

x10user, 2017-11-12:
Hello Andre,

After a couple of years of absence from RC planes (mostly sailplanes) I want to restart this hobby and bought a FrSky X10 to replace my outdated 35 MHz radio gear. I selected this TX with OpenTX in mind. I already flashed it with the available nightly builds and  I'm amazed about it.  I really appreciate the work has been done!

Is there something I can contribute?
I'm an mechanical engineer without deeper experience in programming. But from my work and smaller projects I have still some understanding about it. So maybe I can do some testing (i.e. user interface on the X10 or Companion implementation) or provide drawings, sketches (also 3D) for documentation or do some translation to German language.
Please let me know if such kind of help is needed.

best regards
Danny

Closed #5314.

https://github.com/opentx/opentx/issues/5314#event-1319586763