opentx/opentx #5315: X10S ans crossfire : some telemetry lost but no failsafe

Barbaryann, 2017-10-31:
Hello,
I have some telemetry lost with my X10S and crossfire :
- Crossfire FW 2.11
- OpenTX N417

BF Lua and Kiss Lua works. Slow but works

Thanks

schwabe, 2017-10-31:
Can you check the telemetry RX error counter under statistics?

Barbaryann, 2017-10-31:
I have issue with the statistics mode : https://www.youtube.com/watch?v=Wzt4upfiRhs&list=PLmUct0I1--liC4f3-cGAO_j6Lwbq-y0Ai&index=3
My X10S reboot in HELP MODE :/

schwabe, 2017-10-31:
Can you flash an English version and check if you get the telmetry RX error counter on that?

Barbaryann, 2017-10-31:
Okay, I try...

Barbaryann, 2017-10-31:
Now statistics mode working.
I have More than 10000 on Tlm RX Errs. And it go up and up and up... Now 32789 after 2 minutes...
:/

schwabe, 2017-10-31:
The X10S seems to have the same "feature"as the X7  of not working at high baud rates then.

Barbaryann, 2017-10-31:
Shit !
Sorry.
Now we have to wait somebody who can help to explain the hardware mod to fix this bad FrSky idea... :/

Barbaryann, 2017-10-31:
Thanks for your help

Barbaryann, 2017-10-31:
Somebody can work on the same software resolve issue in the Main/hardware menu tu select lower bitrate ?

closed 2017-11-04 by kilrah

Barbaryann, 2017-11-12:
Hi,
Now it is working when i select lower baudrate (100k).
Somebody can expain what are consequences to be in lower baudrate.
I think latency ?
Does it affect the rc command latency or just the telemetry ?
In which time the latency is bigger ? Four time bigger ?

kilrah, 2017-11-12:
It takes a little more time to get the data to and from the module both for control and telemetry, what the actual impact is would have to be measured with a test setup, likely difficult to predict.

commit 8780339c (2017-11-04): Crossfire baudrate selection for X10 (fix #5315)