opentx/opentx #5317: Getting Global Logical Switches in Radio menu

Tonnie78, 2017-11-01:
I am very pleased with the new function of Global Functions.

I do have a lot of models which all have the same Special Functions and I would like to replace them all by one single Global Function.

The problem I am facing at the moment is that a large number of my Special Functions are activated by a Logical Switch.  So my request is can we have Global Logical Switches. I would prefer to have 32 Logical Switches in the Models menu and the other 32 in the Radio menu.

3djc, 2017-11-01:
Sorry, but this is not possible. Global fonctions cannot refer to  things that might not exist in some models

closed 2017-11-01 by 3djc

Tonnie78, 2017-11-01:
To bad. But thinking about my motor safety switch relies on switch SF and
the Throttle stick which are present in all my models.

I would like to make a global switch L1 a~x  IThr -100 and !Sf^
L2 sticky SF^ L1

I am using this on all my models

Tonnie78, 2017-11-01:
My Request is to have a Radio menu Global Logical Switches. Just like Global Functions. Where I can set Logical Switches that are the same on all models.

I do understand that in order to use them, you need to use data that is available on all models, but isn't that the same for the recently added Global Functions. There also I can activate a function that will not work of the model is not exuipped with the needed hardware. Like take Vario as an example.

schwabe, 2017-11-01:
The global functions don't allow you do use model specific things.

3djc, 2017-11-01:
There is a vario internal function that 1) is always there, 2) does not care if a vario exist or not. that's why you can access this function, not a vario itself (they are two very different thing, albeit they might be looking similar)

Tonnie78, 2017-11-01:
Oké clear. Just thought it would be a good additional feature to make to Global Functions more usefull

kilrah, 2017-11-01:
#2409