opentx/opentx #5320: HORUS (Opentx), negative temperature telemetry values are displayed incorrectly

vspeak, 2017-11-01:
In my HORUS (Opentx), negative temperature telemetry values are displayed incorrectly, e.g. -1 ° C is displayed as 64535 ° C. In my opinion, this is due to a faulty data interpretation, instead of an int16 value, a uint16 value is displayed:
0xFFFF -> int16=-1     but uint16= 65535
Is the problem known?

3djc, 2017-11-01:
Instead of speculating on causes, maybe you could give us actual usefull  details, such as where you are reading this value ? (widget ? telem screen ? script ?)

3djc, 2017-11-01:
Seem to work here:

[image]

[image]

vspeak, 2017-11-01:
exactly there.
VERS: opentx-x12s-2.2.0 (85863358)
DATE: 2017-06-22

schwabe, 2017-11-01:
What do you mean with exactly there? IN the telemetry screen or in the main screen?

3djc, 2017-11-01:
Then clearly it works. What is you actual sensor ?

3djc, 2017-11-01:
I just tested with 2.2.0, works ok too, must be your sensor. Please re-open if you have hard evidence that the isue is with OpenTx

closed 2017-11-01 by 3djc

vspeak, 2017-11-01:
we build own sensors.

vspeak, 2017-11-01:
we use temp2 for lua Status messages

3djc, 2017-11-01:
Then you should have a look at what you are sending, because as far as I can tell, it is ok on Opentx  side, as shown above

vspeak, 2017-11-01:
ok, I'll do. next week I an test also with taranis. with taranis it woks well.

3djc, 2017-11-01:
Here with tmp2:

[image]

vspeak, 2017-11-01:
"The sensor reports an int32, so s -1 as int16 is correctly displayed as 65535"  -->> this can be my issue !! Thank you!

vspeak, 2017-11-01:
I have checked the SPort data. Now I'm transmitting the correct int32 values -> and **my** problem is solved -> negative temperature values are correct displayed-> **there was no problem on opentx side**.

3djc, 2017-11-01:
Good news 👍