opentx/opentx #5321: Global Switches

Tonnie78, 2017-11-01:
**I do not understand why Global Switches would not work.**

I do understand that Global Functions can only relate to things that might not exist in some models. Thats why I think adding a Global Switches Function would be needed in combination of Global Functions. Most of the Special Functions I have in my models I have in multiple models so would be ideal to share them between models using the Global Functions, but at the moment this is not possible because most of my special functions are activated using Logical Switches. For instance my Vario sounds. I have a OpenXsensor vario in all my models, which i activate with a local switch to switch of the Vario if the motor runs. I dont want to hear a sound telling me i climp if the motor runs. I know this. 

So by adding a Global Switch I can make this function once and share between models.
In the current state the Global Functions does not offer me anything.

I am very pleased with the new function of Global Functions.

I do have a lot of models which all have the same Special Functions and I would like to replace them all by one single Global Function.

The problem I am facing at the moment is that a large number of my Special Functions are activated by a Logical Switch. So my request is can we have Global Logical Switches. I would prefer to have 32 Logical Switches in the Models menu and the other 32 in the Radio menu.

3djc commented 4 hours ago
Sorry, but this is not possible. Global fonctions cannot refer to things that might not exist in some models

To bad. But thinking about my motor safety switch relies on switch SF and
the Throttle stick which are present in all my models.

I would like to make a global switch L1 a~x  IThr -100 and !Sf^
L2 sticky SF^ L1

I am using this on all my models

3djc, 2017-11-01:
Duplicated #5317

closed 2017-11-01 by 3djc

Tonnie78, 2017-11-01:
Maybe I am entering this request at the wrong place. Could someone tell me where to post a feature request for OpenTx

schwabe, 2017-11-01:
This has been discussed multiple times (also in other issues). And the short answer is that it is very complex to implement and prone to wrong usage and therefore we not going to implement it any time soon.

notlebowski, 2019-01-10:
I can't judge if it's complex to implement or not, but maybe you have a different implementation in mind when you say it's "prone to wrong usage".

I am repeating in all my powered models the same throttle safety and that one is reliable and I prefer to have it as a radio function instead of repeating it in each model.

I am imagining a setup like this

**Radio**
GS1 Sticky(GS08, GS03)
GS3 ( Ele < -95 ) AND GS04 Delay(0.2s)
GS4 ( Rud < -95 ) AND GS05 Delay(0.2s)
GS5 ( Ail > 95 ) AND GS06 Delay(0.2s)
GS6 Thr < -95 Delay(0.1s)
GS7 FM1 AND !GS06
GS8 SF↓ AND GS06 Delay(0.7s)

**Model**
FM1:Safety !GS01

CH3:Thr
    I3:Thr Weight(+100%) NoTrim [Thr]
 := MAX Weight(-100%) Flight mode(FM1:Safety) [Cut]

This would free up a lot of switches in my models and make them much better readable and better to maintain. Over the years I made changes to this throttle safety (this one is not even the latest), but this means some models don't work slightly different because they never changed.