opentx/opentx #5323: Script non executed in global function (Taranis X9D)

pindeslandes, 2017-11-04:
Companion 2.2.0 and companion 2.1.19.

I have this very simple function script:
_local function son()
playFile('/SOUNDS/fr/wizz.wav')
end
local function rien()
end
return {init = son, run = rien}_

If I use it in a **special function** with switch ON, it works fine (plays the sound once at model loading).

If I use it in a **global function** with switch ON, it doesn’t work at all. (I have checked the 'Use global functions' box, of course.)

mpaperno, 2017-11-29:
Global functions can't run ("play") a script... neither init() nor run(), regardless of trigger method.  There is simply no provision for it in the firmware code that I can see.

Two options:
1. Remove that as possible action for global SFs (the current bug).
2. Add code to handle scripts in global SFs (new feature).

The former would be quicker and simpler to test, with almost no possible side-effects.  The latter could be done later if deemed useful.  It is surprising/telling that this issue hasn't come up before (unless it has).

mpaperno, 2017-11-30:
Fixed with #5440

closed 2017-11-30 by mpaperno