opentx/opentx #5324: N427 X10S

buzzzsam, 2017-11-06:
Icon disappear.  Gimbal not working.  (values jumping everywhere)

kilrah, 2017-11-06:
Check your SD card and contents. Post more info. Screenshot, video.,...

buzzzsam, 2017-11-06:
Formatted 32G FAT32 card with v11 content. 

https://www.youtube.com/watch?v=wwRtg7lkCN0&feature=youtu.be

Same with my friend's unit. 
https://www.facebook.com/groups/rotorriot/permalink/917666715054968/?comment_id=917674201720886&notif_id=1509853034624969&notif_t=group_comment_reply

buzzzsam, 2017-11-06:
will record more video on youtube later.

3djc, 2017-11-06:
Have you run stick calibration ? ,Does it look crazy there too ? Is it always ? Triggered by something ?

schwabe, 2017-11-06:
that looks like pwm gimbals misdetected as normal analog ones or completely broken calibration

buzzzsam, 2017-11-06:
I reformated once again, and placed in amber sound file this time.. seems working..  My friend is testing now.. Will see if it works for him with the sound files.

buzzzsam, 2017-11-06:
I am still getting the shutdown -> emergency error.. I am really suspecting the x10s not liking my 32g sandisk extreme card.

3djc, 2017-11-06:
You video has nothing to do with sd cards (content or support),  you should not get jupping channels like that, no matter SD card status

schwabe, 2017-11-06:
can you make a video of the analogs debug screen? (To see which kind of gimbals you have detected)

buzzzsam, 2017-11-06:
My tx is now in working state(except the shutdown bug which is extremely annoying).  Will ask my friend to post if it is still happening to him.

meoster, 2017-11-06:
He still has the problem, can you let him know where the analog debug screen is?

meoster, 2017-11-06:
By the way, I have a different account on my phone.  Sorry for the confusion.

3djc, 2017-11-06:
The shutdown issue is known, so in this thread, let’s focus on the stick issue, if it can be duplicated

aiyakim, 2017-11-06:
I dont find analog debug screen on my x10s

kilrah, 2017-11-08:
long press enter on home screen->statistics->3rd page

bsongis, 2017-11-12:
anybody else has this issue? I can't reproduce

meoster, 2017-11-12:
My friend and myself seems to have gotten it working after. Don’t know the
reason..

kilrah, 2017-11-12:
Probably was SD card issue.

closed 2017-11-12 by kilrah